ROBOTIS-GIT/DynamixelSDK #1: I got this problem in MatLab ...dxl/DXL_SDK_LINUX_v1_01/src/dynamixel.so: clase ELF errónea: ELFCLASS32

marisoru, 2016-05-07:
Hi! I'm trying to run the dynamixel motors in MatLab on Linux so I download the SDK from this link support.robotis.com/en/baggage_files/dynamixel_sdk/dxl_sdk_linux_v1_01.zip

But when I try to load the library I get this error:
...dxl/DXL_SDK_LINUX_v1_01/src/dynamixel.so: clase ELF errónea: ELFCLASS32  

What I understand is that this is only available for 32 bits and mine y 64 bits. 

I have been looking for SDK for 64 bits but I don't fin it. Does someone knows where I can get it or maybe how to re compile this version. Or any other solution?

Thank you!!!

ROBOTIS-Leon, 2016-05-07:
Hi,

As you said, you need 64bit-compiled library to use it in MATLAB 64bit. 

1. Try edit `MAKEFILE` in `\src`:
CFLAGS		= $(INCLUDEDIRS) -W -Wall -O2
CFLAGS          += -m64

2. To use this option, packages shown below should be installed previously.
`sudo apt-get install gcc-multilib`

I didn't try above yet and I can't say it works, but I will check if it works after holiday period here :)

Furthermore, we are now making MATLAB example using new coming C ver. SDK. I hope it helps you afterwards.

closed 2016-05-16 by ROBOTIS-Leon

HDallali, 2016-07-06:
Hi Leon,
I have recently downloaded the Dynamixel SDK and when I try to run the example read_write.m for a 64 bit windows  in the protocol1.0 folder
if ~libisloaded('dxl_x64_c');
    [notfound, warnings] = loadlibrary('dxl_x64_c', 'dynamixel_sdk.h');
end

I get the following error: 

There was an error loading the library
"...\Documents\DynamixelSDK-master\c\build\win64\output\dxl_x64_c.dll"
The specified module could not be found.

However the file is on the path and in the folder above! I was wondering if you could help me with this?

Thank you,

ROBOTIS-Leon, 2016-07-07:
Hi!

Yes!, you are right!, the library file is in that folder!

but I want to suggest that you see [How to set the file path in MATLAB](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/4.6-MATLAB). It would let your MATLAB know where the library is. :)

If theres some problem, let me know!

HDallali, 2016-07-07:
Hi,
Thank you for your help. Actually the matlab path was set correctly, but I had to rebuild the library files in the C solution with visual studio 2015. I also noticed that all the files in the folder 'DynamixelSDK-master\matlab\' have the 32 bit library. dxl_x86_c. I'm using a 64 bit windows so I changed all 'dxl_x86_c' terms to 'dxl_x64_c' in several folders.  

I think it'll be nice if the matlab files do it automatically, with the computer={PCWIN64 or PCWIN32} command.

Thanks,

HDallali, 2016-07-07:
Hi
The position control example is working. I'm trying to modify read_write for torque control and later impedance control.

Most address values have a lowest and highest bytes. Is there any example showing how to combine the two values? Is there a torque control example?

Thanks,

ROBOTIS-Leon, 2016-07-08:
Hi!

I didn't think that MATLAB can detect the platform version!
Thank you for your suggestion!

There's no example or available functions for value combination, but theres some functions like `DXL_MAKEWORD(a, b)` defined in `packet_handler.h`. I got rid of it from the list of available functions because of the concentration on utility functions for dynamixel operation.

but I see you need that. We will keep it in mind for the next release. :)

SDK example is only for showing on the level of how to change the values on the Dynamixel control table, not for showing its complicated applications. This kind of application examples also would be provided when we notice its importance. 

Thank you!