ROBOTIS-GIT/DynamixelSDK #2: Different return delay time

dkoguciuk, 2016-05-18:
Hello!

I is probably not the problem of your package, but I thought maybe you have already faced similar situation in the past: I have AX-12A dynamixels and using your package and USB2DYNAMIXEL I've got back a frame from servo using Protocol1.0/Ping program, but the time between sent and received package is each time different. I use 1Mb/s and Return Delay Time of 0 (set in EEPROM area). 3 of 4 frames I've got something like 20us of delay:
[image]
But sumetimes I've got really long return delay time (about 150us or even 200us):
[image]

Do you ever seen something like this? Do you have any idea what can cause this problem?
Best,
Daniel

ROBOTIS-Leon, 2016-05-19:
Hi Daniel!
Thank you for your kind explanation of the situation. 
Theres some reason that I assume, but not certain.
I'll deal with this issue asap with the man who are supposed to be charged of.
Sorry for the inconvenience.

ROBOTIS-Leon, 2016-05-24:
Hi! Sorry for my late reply.
I have checked that the team is now figuring out about this issue, and you will get your e-mail soon from cs@robotis.com. 
If you have further questions about our hardwares, cs@robotis.com is always prepared to help you. 
Thank you for your patience and have a good day.

closed 2016-05-24 by ROBOTIS-Leon

toddvarland, 2016-10-01:
This issue doesn't appear to be resolved (that I can tell). Can you please share what the solution was? I too am having this issue (SDK version 3.4.1)

dkoguciuk, 2016-10-02:
Yeah, of course I can share it, but it's no solution at all. I've got an e-mail from Robotis Support Team:

- The Response Time can be affected by the Position/Velocity controller operation. 
- Controller priority is higher than communication protocol priority. 
- Response Time is defined as the time between receipt of the Instruction Packet and transmission of a Status Packet. 
- If the controller is not being used during this time, the Response Time is unaffected. 
- However, if the controller is running before the Status Packet has been transmitted, the Response Time is extended. 
- This is why you have variable Response Times.

So there is no solution to our problem. I write my own program for STM32 dealing with this issue so I can share it if needed.