ROBOTIS-GIT/DynamixelSDK #3: potential out-of-bounds write in Protocol1PacketHandler::WriteTxRx

cclib16, 2016-05-26:
Hello, 

I'm using the C++ sdk. I find that the method `Protocol1PacketHandler::WriteTxRx` only `malloc(length+6)` for txpacket (`src/dynamixel_sdk/Protocol1PacketHandler.cpp:509`), but `rotocol1PacketHandler::TxPacket` writes the checksum into `txpacket[length+6] (length+3+4-1)`, causing a out-of-bounds write.

Instead, the method `Protocol1PacketHandler::WriteTxOnly` calls `malloc(length+7)` (`src/dynamixel_sdk/Protocol1PacketHandler.cpp:485`)

Did you note this difference between the two methods?

Thank you.

ROBOTIS-Leon, 2016-05-27:
Hi,

you're right. It'll be changed into `malloc(length + 7)`. Thank you for your valuable advice!

Your comment will be reflected in next release of the source.

Have a nice weekend.

ROBOTIS-Leon, 2016-06-01:
Hey, how are you.

Your suggestion at issue #3 has been applied to the DynamixelSDK ver. 3.1.0.
You can see the changes.

Thank you.

cclib16, 2016-06-06:
Hi Leon,

Thanks for the quick response!

Just FYI, I also noted that the similar function in the c sdk allocates (length+6) bytes for the txpacket, though I'm not sure whether the same problem happens there.

Thank you.

ROBOTIS-Leon, 2016-06-07:
Hi

Thank you for your continual comment. 

Dynamixel SDK C version source and its manual are waiting for next release, and what you said would be applied today.

I'm very glad to develop MATLAB - SDK C version .DLL binding example with no errors by modifying what you have pointed out.

So thank you and please don't hesitate to ask me further questions if something is going wrong with it.

ROBOTIS-Leon, 2016-06-07:
Dynamixel SDK C version source and its manual is updated as SDK v3.2.0 just now.

Thank you

closed 2016-06-07 by ROBOTIS-Leon