ROBOTIS-GIT/DynamixelSDK #6: DynamixelSDK for ARM like Raspberry Pi

jorgessantos, 2016-06-08:
Any possibility for a ARM version?

ROBOTIS-Leon, 2016-06-08:
Hi,
Sorry for insufficient information on wiki.

We tested Dynamixel SDK in RaspberryPi2B, RaspberryPi3B and it worked fine. 
But during the test, we used ROBOTIS USB2Dynamixel (through /dev/ttyUSBx) as a signal converter.
You can also use its GPIO UART (through /dev/ttyAMAx) for Dynamixel control, but as I know it also needs kinda signal converter. 

and another SBCs using ARM series (Odroid, BeagleBone Black) are planned to be tested with SDK, but seen from my experience, it will work too. 

Any other questions, please don't hesitate and ask me. Thank you.

closed 2016-06-08 by ROBOTIS-Leon

jorgessantos, 2016-06-08:
Hi leon,

Thank you for answer.
I made little changes and works very well. 
I used RX-28 and usb-Rs485 converter on RaspPi 2B.

ROBOTIS-Leon, 2016-06-14:
Nice to hear that!