ROBOTIS-GIT/DynamixelSDK #7: DynamixelSDK for Windows 10 IoT Core

asoaresd, 2016-06-15:
Hi guys!

I'm working with Dynamixel XM430 and USB2DYNAMIXEL.
At first I connect the USB2DYNAMIXEL through Windows x64 and I could run the example ReadWrite, then I connect the USB2DYNAMIXEL through my Raspberry PI 2 running Raspbian LINUX, and worked nicely, I could run the example ReadWrite without problems.

Now I want connect through Raspberry PI 2 running Windows IoT, I debugged and I could run the shell, but I just get Error when opening serial port, I tried several ports, but I don't know exactly where is the error.

Have you guys tried to run the SDK through Windows IoT?

Thanks,
Alex Duarte

ROBOTIS-Leon, 2016-06-16:
Hi, alex. Nice to see you here again

Thank you for your posting.
This kinda application is also valuable for us.

Now, to sum up, you have already tried to open the port with the device name `/dev/ttyUSBx' as linux format or 'COMx' as windows format, right?

I surveyed a little bit more, and it seems like some users use the serial port in RPi2 - WinIOT with same methodology in DynamixelSDK.   

Let me find out the solution, but it may takes couple of days.

asoaresd, 2016-06-16:
Hi Leon!
Yes, you're right, I tried using those formats.
I will try to find out too,
Thank you so much!

ROBOTIS-Leon, 2016-06-20:
Hey alex, did you have a good weekend?

I want to inform that we have no progress here yet. Since WinIOT is very new for us, installations and more stuffs makes me not easily understand about what's going on in the platform. Furthermore, theres no much informations related with the serial port control I found out. I'll keep trying so don't be frustrated :) but it'll takes more time I think, and I just concerned that you may wonder how my survey is going on so I should reply in the first place. 

One thing that makes me interesting is that WinIOT with C++ or C# uses the serial port with automatically detected devices, and controls their paths with kinda `serialPort[0]`, `serialPort[1]`, for example. It seems that they didn't spend time to make user easy to use the serial port.

Any way, If you had never seen this, see 
[https://github.com/ms-iot/samples/tree/develop/SerialSample](https://github.com/ms-iot/samples/tree/develop/SerialSample). I hope this gives you some hints for your recent work.

Leon

asoaresd, 2016-06-20:
Hey Leon, yes I did, how about you?
I would like to thank you for your time! I have tried some modifications but without success, I checked the link you sent to me, I will try using those codes.

Thank you so much!

robotpilot, 2018-05-27:
I am a new person in charge of development and maintenance of DynamixelSDK. This issue has been raised quite a long time ago, but we do not have similar user issues, so we close it arbitrarily. This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2018-05-27 by robotpilot