ROBOTIS-GIT/DynamixelSDK #9: Linux, There is no status packet

2Guz, 2016-06-23:
Hi again,
I am currently using dynamixel sdk develop, and I tried running the ping.c for linux64 and got this ouput.  In addition, when I run the read_write.c file I got this output. 

Ping.c Output
Succeeded to open the port!
Succeeded to change the baudrate!
Now the controller baudrate is : 1000000
[ID:001] Try factoryreset : Aborted
[TxRxResult] There is no status packet!

Read_write.c Output
Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] There is no status packet!
Press any key to continue! (or press ESC to quit!)
[TxRxResult] There is no status packet!
[TxRxResult] There is no status packet!
[ID:001] GoalPos:000  PresPos:000
Press any key to continue! (or press ESC to quit!)
 
I am using the dynamixel2usb if that helps. My question is why isn't the servo I have connected moving by running read_write.c? Also why does it say there is no status packet, and how do I fix this issue?

I have the servo powered with the correct voltage source attached. Thank you for all the help!

ROBOTIS-Leon, 2016-06-24:
Okay
You now can use the SDK.
`[TxRxResult] There is no status packet!` error could be occurred when there are some wrong setting somewhere on the software or hardware. Please check following:

1. [Linux] Run the example source after getting super user authority, which may mean `sudo su`, for example.

2. [USB2Dynamixel] Slide switch on the side of the USB2Dynamixel should be set to:

   - TTL : If you use one of the Dynamixel series with 3P cable
   - RS485 :  If you use one of the Dynamixel series with 4P cable

3. [Power] 12V for all series of the Dynamixel (24V is recommended for PRO series)

4. [Dynamixel] ID and baudnum should be prepared with:

   - AX : ID - 1, baudnum - 1 (1M)
   - PRO : ID - 1, baudnum - 3 (1M)

   (use Dynamixel Wizard for your convenience. It's available at www.robotis.com - download)

5. [Example source] Run AX with protocol1.0 example, PRO with protocol 2.0 example, each.

If the problem looks like not be solved, capture your H/W configuration and results that example shows, so that I can examine them. 

Thank you and keep asking :)

2Guz, 2016-06-24:
Hi again, I have looked online and I think there is no Dynamixel Wizard for Linux. I am also using RX-28 servos. 

I tried sudo su, and still got the same error. Also, I made sure that the slide witch is on RS485 and that the servos are properly powered with 12V, but still got the same error.  I'm sorry, but I don't know how I should do number 4 from your response earlier?

Thank you

closed 2017-08-28 by ROBOTIS-Leon