ROBOTIS-GIT/DynamixelSDK #11: RX-28 Servos: Running 2 Servos

2Guz, 2016-06-27:
I have been able to set the baudrate and the I.D's of both servos to 1 and 2 respectively with the Dynamixel Wizard on Robotplus. I tried running the read_write code in the linux64 directory to make sure that they both were working individually. First I had servo 1 plugged in and ran the code and it moved. After, I changed the id number to 2 in the code, and ran the code with servo 2 plugged in. The result was that the servo moved. 

The strange thing I noticed was that I plugged in servo 1 and servo 2 into a 6 Port RX/EX Power Hub, and ran the code with the id number in the code set to 2 and both servos ended up moving. I'm very confused on why this is happening when I thought the code only runs for 1 servo which has the specific id. Only one of these servos has an id of 2, but why do both of them move?

Here is part of the output in the terminal: 

[ID:001] GoalPos:1023  PresPos:956
[ID:001] GoalPos:1023  PresPos:956
[ID:001] GoalPos:1023  PresPos:956
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:1023  PresPos:000
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:60348
[ID:001] GoalPos:1023  PresPos:1018
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:1023
[ID:001] GoalPos:000  PresPos:956
[ID:001] GoalPos:000  PresPos:1021
[ID:001] GoalPos:000  PresPos:956
[TxRxResult] Incorrect status packet!
[ID:001] GoalPos:000  PresPos:000
Press any key to continue! (or press ESC to quit!)
[ID:001] GoalPos:1023  PresPos:43196
[ID:001] GoalPos:1023  PresPos:1021

Another question I had is why is it stating incorrect status packet? 
Also is there a way to set an inital position for the servo to start at, such as the mid point?
In addition, I ran sync_write and I was wondering what does this code do? 
Lastly, why do you need a Data Byte Length, when in the read_write code you didn't need it. 

Thank you once again for all the help! We were able to at least get our RX-28 servos moving.

ROBOTIS-Leon, 2016-06-28:
Hi!
I'm sorry, I didn't read whole word from your previous comment.

Dynamixel Wizard for Linux is not being serviced now, no schedule to make it, but we are considering about its necessity. To avoid your waste of your time, use one of the example code called `dynamixel monitor` instead, which may let your PC detect Dynamixel connected with or change the ID, baudrate, and stuffs manually.  

Yes, I was wondering why you open new issue again and again :D. You just comment as an overwrite to your previous issue. Just reopen the issue closed previously, the cover would be changed into green. If you have another kind of issue, then opening new issue would be reasonable.

Returning to the subject, 
it worked. 
1. C ver. or C++ ver. - Linux64 - /dev/ttyUSB0/ - protocol1.0 read_write example
2. two Dynamixel connected with eachother, and one of them to USB2Dynamixel. 
3. ID: 1 and 2, set ID in the code to 1 once, and 2 once.
4. Baudrate : default (1M)
5. Power : 12V

I had never used 6 Port RX/EX Power Hub, but no doubt that Trossen Robotics does good work.

Would you show instrument settings? It's for checking whether the instruments are correctly connected. 

In further, I recommend you to try `DynamixelSDK-develop`, not `DynamixelSDK-master`. You can download it by changing the branch tab in the page from master to develop. 

Let's see what's going on on your table :)

2Guz, 2016-06-28:
My partner has a windows computer, so we ran Dynamixel wizard on his computer to update both servos id number and baudrate. We are using the . C ver. - Linux64 - /dev/ttyUSB0/ - protocol1.0 read_write example. We are also using DynamixelSDK-develop. 

[image]
[image]
[image]

ROBOTIS-Leon, 2016-06-29:
Thank you,

It seems that H/W setting is done properly. 

Now, I don't have an idea why it happens because there wasn't any problem on my settings as nearly same as yours (except 6 Port RX/EX Power Hub <-> SNPS2Dynamixel). Take a look [how my stuffs works](https://drive.google.com/folderview?id=0B2-t1GBeX7tuaVI2VVk1MmNJWXM&usp=sharing).

I changed only ID and goalposition range on the code of DynamixelSDK-master which is same one of DynamixelSDK-develop that I told you yesterday, using a couple of Dynamixel RX-28 which are older than yours (so perhaps the firmware version affects nothing) and are directly connected to each other with 4p cable. 

If your codes are very far from the original, I want to see your codes.

Plus, I guess the incorrect status packet error has been occurred by the same reason that occurs kinda `PresPos:60348`.

2Guz, 2016-06-29:
Hi! 
Thank you for all the help. We doubled checked the ID numbers for bot RX-28 servos, and realized that they both had the same ID number although we thought we changed one servo ID to 1 and another to 2. The read_write code is working correctly now.

Some questions I had:
1. Is there a way to change the speed at which the RX-28 servos reach there max positions?
2. Is there any difference between the read_write and sync_write code besides that read_write can control one servo while the sync_write code can control 2? 
3. I'm trying to understand both the sync_write and read_write code, but I don't know where alot of the code is defined such as :tcgetattr", "functl:, or  the type "uint8_t" to name a few. Is there away to see where these things are defined and how do they work?
4.Lastly, we were planning to interface the dynamixel RX-28 servos with an Intel Galileo Gen 2 Microprocessor Board which has a linux part and an Arduino part. Do you have any advice or advice that we can use so we can control the dynamixel RX-28 servos with the Intel Galileo Gen 2 Board?

ROBOTIS-Leon, 2016-06-30:
Well done! You are now a professional. :)

Here is the answer:

1. Officially, the issues related with the thing except SDK have been supported by `support@robotis.com`, so next time I wish you use it. :D. I asked about that to the teammate related with that question, RX can't be controlled like a 'profile control' so the higher layer controller has to calculate it. 

2. You can use either for controlling more than two Dynamixels. To operate two Dynamixels, for example, the normal ReadWrite should be called twice, while the Sync Read/Write is needed to be called only once to transmit signal. Thus, Dynamixels would be operated by time sequence in former case, or would be operated simultaneously in latter case.

3. I didn't use "tcgetattr", "functl" in my code, but perhaps you may see what you wanted in 

http://linux.die.net/man/3/tcgetattr
http://linux.die.net/man/2/fcntl

:)

You can see "uintN_t" in "stdint.h"

4. We don't have any solution for controlling Dynamixel via Intel board yet, but we are planning a schedule for finding out the solutions which uses SBC or low layer devices, such as Intel Edison, Galileo, etc. I guess, from my knowledge, Intel Edison works with yacto Linux, so if it uses serial devices or USB, especially, it will work. This issue may be solved later, but I don't know when it is.

closed 2016-06-30 by ROBOTIS-Leon