ROBOTIS-GIT/DynamixelSDK #15: Failed transmit instruction package

alhatton, 2016-06-30:
Hello,
I keep having an error pop up when I attempt to run the examples in the DynamixelSDK file. I've go a crustcrawler ax-18a smart arm and I'm trying to connect to it with a USB2Dynamixel hooked up to a jetson TK1 board with ubuntu tegra on it. I've tried running the Ping example and it outputs this: Succeeded to open the port!
Succeeded to change the baudrate!
[TxRxResult] Failed transmit instruction packet!
[ID:001] ping Succeeded. Dynamixel model number : 0
I'm still pretty new with UNIX coding and so I'm not sure what the failed transmit instruction packet is or how to fix it. Also I'm not sure if its the problem or not but to get the make file to work I had to comment out the Format: -m64 line. 

Thanks

ROBOTIS-Leon, 2016-07-01:
Hi!

First of all, The notices like "transmit instruction packet" are printed out from the example source code, not from the linux. This may mean that the kernel on the jetson failed to write packet on the buffer. 

-m64 option is to force to build the source in 64bit configuration
There was some issues of -m32 or -m64 option which were occurred by using SBCs.
Keep your work going as you did.

Anyway, did you tested in another SBC? If you have any others, please test with that at first. If there is no error, we should find out any other solution with jetson.
I actually am planning the test with SBCs like jetson, but it has not been done yet.

I'm sorry for the inconvenience

alhatton, 2016-07-01:
I have another jetson, but that is it as far as SBCs go. Is there anyway I can check if it wrote to the buffer or not?

ROBOTIS-Leon, 2016-07-04:
Dynamixel SDK uses `write()` function that is from `fcntl.h` in the code:

```c
int writePortLinux(int port_num, uint8_t *packet, int length)
{
  return write(portData[port_num].socket_fd, packet, length);
}
```

it returns the result of writing so you may check it :)

closed 2016-07-04 by ROBOTIS-Leon