ROBOTIS-GIT/DynamixelSDK #20: How can i install  the SDK on matlab?

leejiawei, 2016-07-19:
Hi : I am unable to install this Dynamixel sdk on Matlab ? Can I have a simple procedures or guidance?
Thanks alot

ROBOTIS-Leon, 2016-07-20:
Hi, mr. Lee

Did you already have a look on [MATLAB guidance for example codes](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/4.6-MATLAB)

If you want an extra support, reply my comment

Thank you

closed 2016-07-20 by ROBOTIS-Leon

leejiawei, 2016-07-20:
thanks man.  i actually haven't seen this before. anyways.  i have one
question . that is. this sdk is for USB2dynamixel right? but my motor
controller is CM700, how can i connect my cm700 with MATLAB interface?
i appreciate your answer
sincerely
jiawei

ROBOTIS-Leon, 2016-07-20:
At first, if you are going to use your CM700 as a signal converter like USB2Dynamixel, there is no F/W on CM700 for signal convert, So you should make functional F/W yourself.
Dynamixel SDK ver.C may help you since it is optimized on buffer management. 

At second, if you are going to implant your MATLAB code on your CM700, it would be better to ask to the MATLAB. I don't have any solutions for it. I'm so sorry