ROBOTIS-GIT/DynamixelSDK #26: Diff between dynamixel_motor

130s, 2016-08-12:
I saw the 1st release of the software from this repo https://github.com/ros/rosdistro/pull/12346 Congratulations!

It'd be great if you could clarify somewhere (e.g. personally I recommend at minimum the description tag in `package.xml` is the best since 1) that's the official package manifest 2) it'll be automatically propagated onto ROS wiki) how this software is different from [dynamixel_motor](http://wiki.ros.org/dynamixel_motor) package, which has been serving as the defacto option in ROS to interact Dynamixel motors (but the maintenance of it has been stale for awhile, so IMHO official package from the manufacturer should be very much welcomed from the community).

It'd be very nice also for the existing users who use Dynamixel via ROS that how APIs are different between yours ant dynamixel_motors. Maybe a table to compare the APIs would work the best.

I don't really know if this package can replace `dynamixel_motors` or if this package is something with completely different purpose/features. But I believe the questions I've raised here are FAQ for ROS users.

Kamsahamnida!

ROBOTIS-Leon, 2016-08-17:
Hello, Mr. Saito.

Heres my reply.

1. Thank you for your advice. We will think about `package.xml` issue.

2. In brief, our official SDK for Dynamixel has some features like:
* Coded with c++
* All Dynamixel models are available in forthcoming ROS example.
* All items or addresses listed on control table are able to be controlled.
* Multi-Port
* GUI type control environment

We will update it on wiki someday.

Arigatou gozaimasu!

130s, 2016-08-17:
Thank you for the prompt response. I'll look forward to the doc update.
As I said, it would be very appreciated by lots of existing users if you could refer in the doc to the [dynamixel_motor](http://wiki.ros.org/dynamixel_motor), which isn't clear yet to me.

130s, 2017-05-17:
There's a similar interest from community like [this one](http://answers.ros.org/question/261887/which-dynamixel-package-shoud-be-used/?answer=261983#post-id-261983).

ROBOTIS-Leon, 2017-05-18:
@130s 

Thank you for notification.
we are thinking about it seriously.

closed 2017-05-23 by ROBOTIS-Leon