ROBOTIS-GIT/DynamixelSDK #33: Bug in XM430 firmware ?

jacqu, 2016-09-23:
Hi,

When using a XM430 in extended position mode, reading present position (register address 132) gives -4096 where position should be -1. Adjacent values are ok (0 and -2).

Position value / Bytes given by a bulk read command
0 -> 0 0 0 0
-1 -> 0 240 255 255 (should be 255 255 255 255)
-2  -> 254 255 255 255

Maybe I missed something. Thanks for helping.
Jacques

jacqu, 2016-09-23:
This is a workaround that seems to work (the Simulink block communicates with a XM430). The glitch happens every multiple of 4096.

[image]

jacqu, 2016-09-23:
I just discovered that it is possible to upgrade the firmware with the R+ manager 2.0. The glitch disappeared in the latest firmware.

Sorry for the noise. You can close the issue.

ROBOTIS-Leon, 2016-09-24:
You did great!

Anyway, the issue which is related with the hardwares including Dynamixels and its firmwares, and stuffs, would be glad if you ask support@robotis.com for next time. :)

closed 2016-09-24 by ROBOTIS-Leon