ROBOTIS-GIT/DynamixelSDK #34: Problem with control of Dynamixel AX-12A by using MatLab

GrigoriyAY, 2016-09-29:
I plan to work with Dynamixel AX-12A, which is connected to computer (Win64) by USB2Dynamixel. I use information from Dynamixel SDK Manual (item 4.6) for adding libraries in MatLab (it's files for DynamixelSDK-3.4.1), but when I run file "read_write.m", I got error:

> Index exceeds matrix dimensions.
> 
> Error in
> loadlibrary>getLoadlibraryCompilerConfiguration
> (line 497)
>         compilerConfiguration=compilerConfiguration(1);
>         %unix machines return c and cpp
>         compilers here
> 
> Error in loadlibrary (line 253)
>     [thunk_build_fn,preprocess_command]=getLoadlibraryCompilerConfiguration(ccinclude,header,headername,compilerConfiguration);
>     
> Error in read_write (line 60)
>     [notfound, warnings] =
>     loadlibrary(lib_name,
>     'dynamixel_sdk.h', 'addheader',
>     'port_handler.h', 'addheader',
>     'packet_handler.h');

Could you help me, what should I do?

GrigoriyAY, 2016-10-14:
It was problem with compiler, which I solved by installing Microsoft Visual Studio Professional 2013. On some reason I couldn't use Microsoft Visual Studio 2015.

ROBOTIS-Leon, 2016-10-18:
Hi Grigoriy
Sorry for my late reply

You've tried in a correct way, that's so thankful.

But I will make your computer be able to use with MS VS 2015 built file.
I should edit something in the source.
Please try and let me know if modified source works after all :)

ROBOTIS-Leon, 2016-10-19:
I just uploaded modified source on the develop branch.
please let me know whether it solves the problem you told

GrigoriyAY, 2016-10-19:
Leon, thank you, but I will try your solution later. Do you know, it is possible to control in MatLab Dynamixel motor which is connected to computer by using Arduino?

ROBOTIS-Leon, 2016-10-20:
Hi

If you are going to use your arduino board as a signal converter that makes TTL/RS485 signal,
I don't know about how to port the arduino with your Laptop in MATLAB, but yes you can do.

When you are going to use it as a Dynamixel controller, and it just gets the order from your Laptop, there is no Dynamixel Library that can be used with Arduino IDE, so you can't perhaps, 
or make the Dynamixel Library yourself by referring the source built for OpenCM IDE (See at support.robotis.com/en), then it would be OK.

Plus, I am now working on developing new Dynamixel Library for ArduinoIDE using my DynamixelSDK. So, if you are interested in, how about waiting nearly 2 weeks? or if you are in urgent, keep doing above.

closed 2016-10-20 by ROBOTIS-Leon

GrigoriyAY, 2016-10-20:
Thanks a lot! Sure, I can wait.

ROBOTIS-Leon, 2016-11-04:
Hi,

I just finished to make the DynamixelSDK for OpenCR, which is a next version of OpenCM but can be used with ROS serial in ROS 2.0. Anyway, the OpenCR is also able to be used as Arduino Board but you can just think that this is kinda Arduino board specialized for Dynamixel.

Send me e-mail to rwjung@robotis.com so that I can send you the file.
We'll upload onto the GIthub when all the work is finished

I hope it can work well :)