ROBOTIS-GIT/DynamixelSDK #36: Problem using MatLab with MX-28T

DannyDRG, 2016-10-18:
I am having trouble using matlab with the USB2Dynamixel and the MX-28T. I am using windows 10 and i have followed the steps to compile the library using Microsoft Visual Studio 2015 Express for Desktop. I have set all the paths according to the available tutorial. Unfortunately when i run the read_write.m file, i receive the following errors. 

Index exceeds matrix dimensions.

Error in loadlibrary>getLoadlibraryCompilerConfiguration (line 497)
        compilerConfiguration=compilerConfiguration(1);  %unix machines return c and cpp compilers here

Error in loadlibrary (line 253)
    [thunk_build_fn,preprocess_command]=getLoadlibraryCompilerConfiguration(ccinclude,header,headername,compilerConfiguration);
    
Error in read_write (line 60)
    [notfound, warnings] = loadlibrary(lib_name, 'dynamixel_sdk.h', 'addheader', 'port_handler.h',
    'addheader', 'packet_handler.h');

I am using Matlab 2014a. I appreciate any help you are able to provide.

ROBOTIS-Leon, 2016-10-19:
Hi, Danny

#34 also told about nearly the same issue
so how about doing like that at first?

and I uploaded modified source on the develop branch.
please let me know whether it solves the problem you told

DannyDRG, 2016-10-19:
Yea, i was able to solve the issue using the same method. Thank you Leon. 

By any chance, do you know if its possible to use the CM-700 controller with Matlab or if the Arbotix-M controller would be the better route? I am attempting to control a MX-28T servo using a joystick for our project.

ROBOTIS-Leon, 2016-10-20:
Hi, Danny

See #20 :)
but if you still have question, feel free to ask.

closed 2016-10-20 by ROBOTIS-Leon