ROBOTIS-GIT/DynamixelSDK #38: Using the USB2AX with DynamixelSDK

abencomo, 2016-11-01:
Can the DynamixelSDK work with the [USB2AX](http://xevelabs.com/doku.php?id=product:usb2ax:faq#qhow_to_power_the_servos) device instead of using the the USB2Dynamixel?

ROBOTIS-Leon, 2016-11-02:
Hi!

I'm not sure that it can work, but

I think that USB2AX must be built for using AX series, so perhaps it would work in TTL interface

abencomo, 2016-11-02:
Hi Leon,

I just tested it using the ping.cpp and read_write.cpp examples and it seems to work.  However, do you know why they suggested to replace the dxl_hal.c with the this [one](https://raw.githubusercontent.com/Xevel/usb2ax/master/pc_software/usb2ax_DynamixelSDK/linux_compatibility/dxl_hal.c)?

I don't think DynamixelSDK uses that dxl_hal.c file anymore since I cannot find it in the repository.

ROBOTIS-Leon, 2016-11-02:
Hi,

To make short, you don't need to understand about what you mentioned, if you are going to use DynamixelSDK 3.0.
[Here](https://github.com/Xevel/usb2ax/tree/master/pc_software/usb2ax_DynamixelSDK/DynamixelSDK_sync/src) uses DynamixelSDK 1.0 for Linux.

The functions in each source file are migrated to the new version as shown below:
dxl_hal.* -> port_handler.*
dynamixel.* -> packet_handler.* and the others.

In usb2ac_DynamixelSDK, they modified some codes in dxl_hal.c to use `"/dev/ttyACM*"` port which recognizes the usb2ax device. 

`sprintf(dev_name, "/dev/ttyUSB%d", deviceIndex);`

to 

`sprintf(dev_name, "/dev/ttyACM%d", deviceIndex); // USB2AX is ttyACM`

In the DynamixelSDK 3.0, if your usb2ax is still recognized as ttyACM instead of ttyUSB,
the DynamixelSDK 3.0 c version read_write example, for instance, edit following codes.

`portHandler("/dev/ttyUSB0");`

to 

`portHandler("/dev/ttyACM0");`

You don't need to change the library source as they did in previous DynamixelSDKs.

abencomo, 2016-11-02:
Great!  Thanks again Leon for the quick response.  I'll close this now.

closed 2016-11-02 by abencomo

ROBOTIS-Leon, 2016-11-02:
My pleasure :)