ROBOTIS-GIT/DynamixelSDK #39: Setting the unique ID of each actuator

abencomo, 2016-11-08:
Does the DynamixelSDK have a function that allows to set  the unique ID of each actuator or do I need to use DYNAMIXEL Wizard for that?

ROBOTIS-Leon, 2016-11-08:
Hi,

Could you tell me more about what 'Unique ID' is?

The following are what I guessed.

1. Theres no any specific function that changes ID is not in the DynamixelSDK.

2. 'I want to set the ID by myself anyway' : 
  visit [support.robotis.com/en](support.robotis.com/en) and see the control table your Dynamixel has.
  Find the address of ID, and change its ID by write1ByteTx() or write1ByteTxRx();

3. If you are a ROS user, try using [Dynamixel Workbench](https://github.com/ROBOTIS-GIT/dynamixel-workbench) that supports you with GUI environment.

4. 'I want to set the ID to 0 or over 255':
You can just change it by following step 2., but it wouldn't work.

abencomo, 2016-11-08:
Hi Leon,

Answer 2 is what I was looking for.  I'd like to change the ID of each servo programmatically instead of using the DYNAMIXEL Wizard. Thanks!

[image]

abencomo, 2016-11-08:
Leon,

How can I persist/apply the changes to address 25 (LED)?  I keep changing it via the DYNAMIXEL Wizard, but I after I power down and power up the servo again, the LED is Off (value is 0).

[image]

ROBOTIS-Leon, 2016-11-08:
Hi,
I hope you might become able to change the ID :)

the Torque Enable and LED value would be initialized to 0 automatically, when you shut its power down.

Note that those so-called Items placed in the RAM can't be kept its data when the power is gone.
Only the red stuffs shown in previous picture will be kept permanently.

closed 2016-11-08 by abencomo