ROBOTIS-GIT/DynamixelSDK #44: hi, i wanna control a servo motor using labview. give me some tip

pasplis, 2017-01-10:
안녕하세요 랩뷰를 이용해서 다이나믹 셀 서보모터를 제어해보고자 합니다.

깃허브 위키에서 제공하는 예제에 랩뷰를 이용한 서보모터 제어 블록다이어그램  화면을 봤는데요.

VI 파일이 제공되어있지 않아서 화면만 봐서는 잘 이해가 되지 않네요.

코드 카테고리에서 이것저것 다운받아서 한번 코드를 참고해보고자 했는데 제가 가지고 있는 랩뷰 버전보다 상위 버전에서 만들어진 파일이라 열어보는게 불가능 했습니다. 

제가 가지고 있는 버전은 랩뷰2011 버전이에요.

혹시  제가 가진 버전에서 열어볼 수 있는 랩뷰 예제 VI 파일을 얻을 수 있을까요?

ROBOTIS-Leon, 2017-01-10:
[Please note] Q&A in Korean is being serviced by contacting with support@robotis.com or Q&A bulletin board.

[The question] was that the version of being used LabVIEW is lower than the one that actually built the DynamixelSDK LabVIEW files, and needs some guide to use it. 

[Reply] Sorry for the inconvenience. We will update the manual of how to use DynamixelSDK LabVIEW sooner. I can't do the LabVIEW Q&A now, but I can say that the DynamixelSDK LabVIEW examples you've downloaded are almost as same as the examples which were developed with the other languages, such as C, MATLAB, etc., and nothing special. 
If you've succeeded to make your .lvlib (by importing the c version .dll file), just connect the nodes which is shown in the Wiki you said.

pasplis, 2017-01-10:
Thank you for replying.

I will try as you say!

closed 2017-01-10 by pasplis