ROBOTIS-GIT/DynamixelSDK #45: port conneting problem

shyeonaaa, 2017-01-11:
Hello! It's good material. thank you for making it.

anyway I'm using MATLAB R2016a on windows 10 and using Dynamixel MX-28.

I have a problem that failed to open the port with matlab.

whenever I run this program, it doesn't work with "Failed to open the port!"

however the dynamixel is normally connected to Dynamixel Wizard Roboplus.

I want know why it doesn't work.

if you let me know, I will really appreciate it.

ROBOTIS-Leon, 2017-01-11:
Hi, 

Sorry for the inconvenience.

Would you check if your com port number is over 10?

shyeonaaa, 2017-01-11:
Hello!

My com port number is 10. Is it any problem?

ROBOTIS-Leon, 2017-01-11:
Ok, let's try to change the com port number to under then 10 through the device manager -> select the com port which points out your USB device , then go properties -> port settings -> advanced settings -> select another com port number, then click ok.

It must be worked well.

closed 2017-01-12 by ROBOTIS-Leon