ROBOTIS-GIT/DynamixelSDK #46: Bulk_read with AX series

KarolloS, 2017-01-13:
Referring to the [documentation](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/2.2-Dynamixel), the bulk_read function doesn't work with the AX series. However in protocol1.0 [folder](https://github.com/ROBOTIS-GIT/DynamixelSDK/tree/master/matlab/protocol1.0) there is a bulk_read example what may suggest that it should work. Which information is true?  

I’m trying to control over 20 servos from MatLab. Read_write functions work fine (except printTxRxResult and printRxPacketError functions, I also don’t know why) , sync_write also, but when I’m running bulk_read example I’m getting communication error (-3001 -> [TxRxResult] There is no status packet!) as a result of function write2ByteTxRx. I was trying to rebuilt library but it doesn’t help.

Anyone has the same problem and found any solution? Or maybe library is not supporting bulk functions with AX servos?

ROBOTIS-Leon, 2017-01-14:
Hi, 

Sorry for the inconvenience.

Here is the reply.
1. As you noticed, the bulkread function doesn't work with the AX seires Dynamixel.
2. the [bulkread.m](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/matlab/protocol1.0/bulk_read.m) you've pointed out tells that
'% Available Dynamixel model on this example : MX or X series set to Protocol 1.0', which means at least the example can work with the MX or X series DynamixelSDK. But I see the needs that the issue related with the BulkRead function should be noticed onto the document, thanks for your pointing.
3. Unfortunately, while using the AX series, you can't get the information from a number of Dynamixels at the same time... only sync write is available. Please note the informations shown in [http://support.robotis.com/en/](http://support.robotis.com/en/) -> Product Information -> Actuator -> Communication 1.0 -> Kind of Instruction -> find BULK READ
4. printTxRxResult and printRxPacketError function is not being serviced in the DynamixelSDK except c and c++ version. Do you think that you strongly need those functions?

KarolloS, 2017-01-15:
Thank you for comprehensive explanation, now I understand everything. So the fact that bulkread is         not supported by AX series is a hardware issue or simply software is not yet prepared? Do you know if someone is working on it?

In case of printTxRxResult and printRxPacketError function  -> of course they are not necessary but usually it is easier to notice error when you can check it in software than observing LED’s

ROBOTIS-Leon, 2017-01-16:
1. Yes, it's a hardware issue. The flash memory in the AX is not enough to carry on the Bulks. 

2. OKay, I will take it into account.

KarolloS, 2017-01-16:
Now everything is clear. Thank you for very quick answer!

closed 2017-01-16 by ROBOTIS-Leon