ROBOTIS-GIT/DynamixelSDK #47: Problem with USB2Dynamixel

GrigoriyAY, 2017-01-26:
I used DC-DC LM2596 for power supply of motor MX-28T, perhaps one of the wires is disconnected and my computer is switched off. Thereafter USB2DYNAMIXEL ceased to detect the presence of motors via program RoboPlus. I tried to connect other motors, but nothing has changed. Could you tell me, it is possible to fix adapter USB2DYNAMIXEL or do I have to buy a new one?

ROBOTIS-Leon, 2017-01-26:
Hi, nice to see you again

To get more accurate information, please ask cs@robotis.com with all details you have, since here only deal with the DynamixelSDK but not the hardware issues :D

In my experience, it seems like your power supply made countercurrent on the system, and it affected to the USB2Dynamixel, crushed the communication chipset. If it is as I said, you should buy another one.
Check if the slide button is on the TTL just to be sure.

And I strongly suggest you to put kinda USB hub between the USB2Dynamixel and your computer to avoid the direct damage in your computer from unintended electrical issues

GrigoriyAY, 2017-01-27:
Thank you for the quick response and the address of the technical support!

closed 2017-01-28 by ROBOTIS-Leon