ROBOTIS-GIT/DynamixelSDK #53: SDK protocol 1.0 MX-64. read2ByteTxRx fails with Comm_RX_TIMEOUT or COMM_RX_CORRUPT

91xxiong, 2017-02-14:
Hi, 

I am using the SDK matlab code to control a MX-64. When I use read2ByteTXRX to read out the present position, it usually takes 1ms for a round trip. But occasionally it fails and takes 35ms with error code either COMM_RX_TIMEOUT or COMM_RX_Corrupt. Do you know how to fix this problem to make it reliable?

ROBOTIS-Leon, 2017-02-14:
Hi,
sorry for the inconvenience.

I need more details kinda environments, devices, hardware settings, software source code, if you don't mean, as much as you can give me, pictures,...

Occasional communication errors between the computer(or micro controller) and the Dynamixel can be occurred in various situations.

What kind of device are you using now?

91xxiong, 2017-02-14:
Thank you for your reply Leon. I use USB2dynamixel to command the MX64 from Matlab Win7. The power comes separately from a power supply. I am able to use sync_write to command 6 motor simultaneously. But when I try to read one motor position, it started to fail. The code is like this,

for loop
    groupsyncwriteAddParam...
    groupSyncWritePacket(group_num)
    pause(0.005);
    read2ByteTxRx(...) % only read one motor here
    readResult = getLastTxRxResult(..) 
    pause(0.005);
end loop

The readResult sometimes is -3001 or -3002. 

If I don't syncwrite, then the readresult is always success. If I syncwrite more motors, the readresult fails more frequently.

ROBOTIS-Leon, 2017-02-15:
Okay, I will see what happens here and tell you what to do :)

ROBOTIS-Leon, 2017-02-15:
Hello,
We will find the reason out as soon as possible.
By the way, if you want to do a simultaneous control, we everytime recommend user to use the sync/bulk functions, and in your case, the bulkread function is better than the normal read function.
See how the Dynamixels works in the Bulkread example.

91xxiong, 2017-02-15:
Thank you Leon. I tried Blukread before read2ByteTxRX. Same thing happened. Meanwhile I will try other things. I will also let you know if I find the problem. Thanks.

91xxiong, 2017-02-15:
Hi Leon, I happened to try my code on Ax12 motors. They worked with no problem. Would there be any fundamental differences between Ax12 and MX64 with respect to the reading?

ROBOTIS-Leon, 2017-02-17:
Hi,

I made the environments as nearly same as possible, and there was no results that you described.

* WIN10
* MATLAB 2016, but the source was developed in MATLAB 2012
* DynamixelSDK which is available now
* Fives of MX-64 Dynamixels
* 12V-5A
* USB2Dynamixel

heres [film and source code](https://drive.google.com/open?id=0B2-t1GBeX7tuWUhvZjk4VDQ0Z1k)

so check this and give me feedbacks :)

and,

basically the AX series and the MX series respond with same ones by read instruction packets. I think the MX that you have doesn't work well. Check the hardware status through the [RoboPlus](http://en.robotis.com/BlueAD/board.php?bbs_id=downloads&mode=view&bbs_no=1132559&page=1&key=bbs_subject&keyword=roboplus&sort=&scate=)once again.

91xxiong, 2017-02-17:
Thank you for your testing Leon! Based on your success, I started to trust the underlying code. So I checked the hardware very carefully. When I disconnected the 6th motor, the reading succeeded for the first 5 motors. Then I figured it was the problem with the wiring length. After I split the power and also the data wire from usb2dynamixel into two strands, one to ID 1,2,3 and the other to 4,5,6, everything works. Now I can use bulkread to read 6 motors within around 4 ms. 

Thank you, appreciate your help.

ROBOTIS-Leon, 2017-02-18:
Okay, it sounds like the issue was in the powering.
I'll try to connect more Dynamixels and check whether it makes the same problem.

You caught a great point. I will take updating this idea on the FAQ bulletin into account.

Thanks for all your contributions. :)

closed 2017-02-18 by ROBOTIS-Leon