ROBOTIS-GIT/DynamixelSDK #57: error by following tutorial example

jugal94, 2017-02-23:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?
Dynamixel X
  [　-　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?
USB2Dynamixel
  [　　　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?
3.4.3
  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　　　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)
I am using the example as shown in the tutorial to operate DynamixelX motor using Ubuntu 14.04, everything is going right until the last command to load the c++ example. it is returning : THERE IS NO STATUS PACKET!

ROBOTIS-Leon, 2017-02-24:
Hi, 

Sorry for the inconvenience.

Let me check something:
1. From which example did you find the problem out 
2. is the slide switch in the left side of the USB2Dynamixel set to the RS485?
3. Did you edit the source code of the c++ example? : ex) Your Dynamixel ID / ADDR numbers or LENGTH of the Dynamixel Control Table, because protocol1 example runs the MX series, and protocol2 example runs the PRO series as default. 
4. If 2, 3 were already done, I need more details, since the issue you told can be occurred in various circumstances. Taking picture of the developing environment and let me know about 1. will be effective.

jugal94, 2017-02-24:
Hello,

Thank you for your prompt reply. I will try to answer your points best I
can.

1. I tried examples as instructed in the quick start tutorial. I tried both
examples 1 and 2, I tried using different motors, different CPU. The
problem still persists.

2 As instructed I used RS485 as I was using 4p cables.

3 I haven't edited the C++ code, but I am using Dynamixel X series motor. I
will check the code and get back to you.

Thank you,
Jugal

ROBOTIS-Leon, 2017-02-24:
Okay, great.

Please check whether you did `sudo chmod a+rw /dev/ttyUSB0` or any other port number to get port access authority.

and see [http://support.robotis.com/en/](http://support.robotis.com/en/) -> <kbd>Product Information</kbd> -> <kbd>Actuator</kbd> -> <kbd> DynamixelX </kbd> -> <kbd> Dynamixel model </kbd> and change the ADDR & LENGTH number by referring the Table.

ex)

// Control table address
#define ADDR_PRO_TORQUE_ENABLE          64 // from 562     
#define ADDR_PRO_GOAL_POSITION            116 // from 596
#define ADDR_PRO_PRESENT_POSITION       132 // from 611

jugal94, 2017-03-02:
> and see http://support.robotis.com/en/ -> Product Information -> Actuator
> -> DynamixelX -> Dynamixel model and change the ADDR & LENGTH number by

Hi Leon,

I am still getting the same error. "There is no Status Packet".
I have ordered 6 Motors from ROBOTIS:

1. Dynamixel PRO L42-10-S300-R (x1)

2. Dynamixel X XH430-W350-R (x4)

3. Dynamixel AZ-12A (for gripper)

I am able to run all of these in R+ Manager. But none of them in Linux
environment. I am giving a step by step description of all my commands so
we can sort out the error quickly.

currently I am using the Dynamixel PRO L42 for my example test.

*1.* The example is from protocol 2.0 as I am using Dynamixel PRO.
*2.* This motor is running perfectly with R+ manager, I have used all the
settings for the linux environment as well. including USB2Dynamixel to RS485
*3*. I have edited the source code and am attaching the commands here
#define DXL_ID                          1                   // Dynamixel
ID: 1
#define BAUDRATE                        1000000
#define DEVICENAME                      "/dev/ttyUSB0"      // Check which
port is being used on your controller
                                                            // ex) Windows:
"COM1"   Linux: "/dev/ttyUSB0"

#define TORQUE_ENABLE                   1                   // Value for
enabling the torque
#define TORQUE_DISABLE                  0                   // Value for
disabling the torque
#define DXL_MINIMUM_POSITION_VALUE      -600            // Dynamixel will
rotate between this value
#define DXL_MAXIMUM_POSITION_VALUE     600            // and this value
(note that the Dynamixel would not move when the position value is out of
movable range. Check e-manual about the range of the Dynamixel you use.)
#define DXL_MOVING_STATUS_THRESHOLD    20                 // Dynamixel
moving status threshold

#define ESC_ASCII_VALUE                 0x1b

*4. * I have attached a photo of the developing environment.
*5.* I am using Ubuntu 14.04

I hope to hear soon from you. Thank you for all your help and co-operation.

Regards,
Jugal Buddhadev

ROBOTIS-Leon, 2017-03-03:
Hi, again

1) 3. Dynamixel AX-12A does TTL communication 3p, so you must be using two USB2Dynamixel, right? the PRO and XH series use RS485 comm 4p. 
2) Do all the Dynamixels work when you connect just one... each of them without any problem?
3) *3.* Dynamixel PRO L42 moves in the range from 0 to 4095. The other PRO can move to minus value position. 
See the details of the Dynamixels in http://support.robotis.com/en/ -> <kbd>Product Information</kbd> -> <kbd>Actuator</kbd>
 Maybe I should write this somewhere clearly, sorry for giving an insufficient information.
4) *4.* Sorry, I can 't find any picture here :v

Please check 2), so that we can know what the problem is.
If still your Dynamixels don't work, let me see the part of your source code which would actually affects to the Dynamixel control by sharing google drive personally. My e-mail address is available in my github page.

jugal94, 2017-03-03:
>       1. Dynamixel AX-12A does TTL communication 3p, so you must be using
>    See the details of the Dynamixels in http://support.robotis.com/en/ -> Product

Hello Leon,

1.  I am aware of the different communications channels for AX-12A, PRO and
XH motors.
2. The motors do not work in any manner of connection (except by using R+
manager, where they run in all possible settings)
3. I am using the advisable range as given in the control table.
4. I have reattached the image.

I will share image of the source code on Tuesday.

Thank You for all your help and advice.

Regards,
Jugal Buddhadev

ROBOTIS-Leon, 2017-03-06:
Im sorry, i cant find anywhere what you attached
Please send me via email. You can find in at my page.

Okay no problem, I will wait for it :)

jugal94, 2017-03-09:
Hi Leon,

I was finally able to run my motors. The error was due to the fact that I
was trying to start the motors from 0 degrees (out of 360 degree) and it
wouldn't work. When I gave it goal position 0 (out of 4095) using R+
Manager, and then on retrying the command. It worked. Thank you for all
your help. You can close this issue now.

Regards,
Jugal

ROBOTIS-Leon, 2017-03-09:
Aha, 

great work

I will think on this

closed 2017-03-09 by ROBOTIS-Leon

akashgupta12345, 2018-10-23:
I changed 
 DXL_MINIMUM_POSITION_VALUE      = -150000;     
DXL_MAXIMUM_POSITION_VALUE      = 150000;

to less values. 

It shows Dynamixel has been successfully connected.

The motors are not moving. The motors work with R + motion when goal motion is changed.

kijongGil, 2018-10-24:
Hi @akashgupta12345 :)
Could you tell me what DXL you used?