ROBOTIS-GIT/DynamixelSDK #58: Dynamixel SDK for Phar lap ETS operatiing system (National Instruments PXI system)

chintals, 2017-02-23:
Hi Guys, 
I am having issues with library import to a National Instruments PXI system running on Phar lap ETS operating system. On a quick compatibility check using a DLL checker from NI yielded 5 bad imports and 18 stub imports: http://digital.ni.com/public.nsf/websearch/0BF52E6FAC0BF9C286256EDB00015230?OpenDocument
The intent is to use LabVIEW code deployed on the Real_time target and let the PXI 8840 controller control the servo's.

Any guidance as to how to proceed to fix the errors related to bad import and stub imports (see figures attached below)?

The Servo is connected to the RS-485 module, which is connected to the PXIe chassis 1078. The PXI 8840 Quadcore Real-Time controls the modules in the chassis. The PXI 8840 run on a Real Time Phar lap ETS operating system. 

Any help or guidance with regards to this problem is much appreciated, 
Thank you, 
Sunil

Please fill out the questionnaire to give you the best support service.  
- Which DYNAMIXEL is it?
DYNAMIXEL PRO H54-200-S500-R
- Which CONTROLLER is it?
PXI 8840 RT via RS485 Connection
- Which VERSION of DynamixelSDK is it?
Version 3, Protocol 2
- Any PICTURES or VIDEOS?
[image]
[image]

ROBOTIS-Leon, 2017-02-26:
Hi, 

I'm not professional in NI products at all, 
but it perhaps shows the function which is being used on the DynamixelSDK APIs but can't be imported by the NI system.

Now I can recommend you only to inactivate those functions by commenting or erasing following stuffs
at <kbd>c</kbd>→<kbd>src</kbd>→<kbd>dynamixel_sdk_windows</kbd>→<kbd>port_handler_windows.c</kbd>→ 259~266 th line and 265~266 th line:

```c
  //if (SetCommMask(portData[port_num].serial_handle, 0) == FALSE) // Not using Comm event
  //  goto DXL_HAL_OPEN_ERROR;
  //if (SetupComm(portData[port_num].serial_handle, 4096, 4096) == FALSE) // Buffer size (Rx,Tx)
  //  goto DXL_HAL_OPEN_ERROR;
  if (PurgeComm(portData[port_num].serial_handle, PURGE_TXABORT | PURGE_TXCLEAR | PURGE_RXABORT | PURGE_RXCLEAR) == FALSE) // Clear buffer
    goto DXL_HAL_OPEN_ERROR;
  // if (ClearCommError(portData[port_num].serial_handle, &dwError, NULL) == FALSE)
  //  goto DXL_HAL_OPEN_ERROR;
```

then, follow [instructions for library building](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/3.2.1.1-C-Windows) to build the dxl_x86_c.dll again

ROBOTIS-Leon, 2017-04-13:
@chintals 
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-04-13 by ROBOTIS-Leon

ddrizzle, 2017-11-02:
I was able to go through the Instruction Packet Document to construct Protocol 2 binary packets and then built my own LabVIEW device library using VISA commands.  There are some subtle details that I came across that you should watch out if you are going to try to go that route.

Jullambert, 2021-08-04:
Hi @ddrizzle 

Actually, I'm having similar issues as @chintals.
I've tried a similar approach as yours (building my own functions (such as Led ON/OFF, Torque ON/OFF and set Goal Position values) for Labview 2012).
They work fine on a regular Windows computer but I'm having issues as soon as I use them on the RT target (it's a regular computer that we converted into an RT controller with Phar Lap installed).

Would it be possible to discuss together the "subtle details" that you came across?

Thank you in advance,
Best regards,