ROBOTIS-GIT/DynamixelSDK #59: Very slow dynamixel read

hvdbrand, 2017-02-27:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it? MX64,MX106
- Which CONTROLLER is it? Ubuntu 16.04 via USB2Dynamixel
- Which VERSION of DynamixelSDK is it? 3.4.3 in  Ros kinetic

I have checkout the DynamixelSDK master branch. (last release note is dated 02.17.2017).
I copied the ros directory into my catkin workspace and ran catkin_make.

I am trying to control a set of dynamixel motors at 100 Hz.
Writing at that rate is no problem.

However, reading values is very slow.
Reading three fields (goal_position, present_position, present_load) takes 7 ms.
Therefore I have reduced the read rate to 10 Hz.

I use a read function which calls the read2ByteTxRx method of the packet_handler.
I use protocol version 1.0.

I can imagine that I am making a mistake somewhere.
Could anyone indicate what I might be doing wrong?

ROBOTIS-Leon, 2017-02-28:
Hi,

1. set the baudrate over than @ 2M (2,000,000) bps
2. use bulkread function, when you use multiple Dynamixels and use read function

hvdbrand, 2017-02-28:
Thank you for your response.
ad 1.) I have some issues in receiving the packet for a baud rate of 1 M bps. So I would have to look into that first.
ad 2.) Blukread would help to a set parameters from different ids. However, it does not speed up reading multiple field from a single motor. 

I have looked at this issue some more and expect this is due to 1 ms latency introduced by the ftdi chip in the USB2Dynamixel interface.
This wait time is introduced because the status packet is too small to fill the buffer on the ftdi.

The ftdi could also trigger on a specific event character that signals the end of a message.
Therefore, on revision of the RS485 protocol it could improve performs if an event character is append after the checksum.

ROBOTIS-Leon, 2017-03-01:
1. Okay.
2. Nope, I mean, if you use multiple Dynamixels and you want to read several items, you can use the Bulkread as like a Syncread. It will speed up.

for 3 MX with 3 by 2 Byte items
ex for read2ByteTxRx)

read ID 1 goalpos
read ID 1 prespos
read ID 1 presload
read ID 2 goalpos
read ID 2 prespos
read ID 2 presload
read ID 3 goalpos
read ID 3 prespos
read ID 3 presload

8 bytes x 9 instruction/status packet
9 times of hardware depended delays

ex for bulkread)
bulkread ID 1 2 3 goalpos
bulkread ID 1 2 3 prespos
bulkread ID 1 2 3 presload

14 bytes x 3 instruction/status packet
3 times of hardware depended delays

If it depends on your coding style or the function structure you planned, I will not touch it but obviously...

hvdbrand, 2017-03-01:
Thank you very much for suggesting the bulkread.
In your example you show a reduction in hardware delays from 9 times to 3 times.
This could even be reduced to 1 hardware delay by making bulkread read in one go from goal position up to and including the present load.
This does solve my issues at the moment.

However, if you ever revise the communication protocol, I would include an event_char after the checksum. This way the FTDI chip does not have to wait for 1 ms before it sends its data.

I consider my issue resolved. Thanks for your fast response and helpful suggestions.

closed 2017-03-01 by hvdbrand

ROBOTIS-Leon, 2017-03-01:
I would like to consider your suggestion for the next SDK release.
Thank you for your new idea. :)