ROBOTIS-GIT/DynamixelSDK #60: Dynamixel MX-28 does not go to the exact goal position

pguogithub, 2017-03-08:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　MX-28R ]  
- Which CONTROLLER is it?

  [　USB2Dynamixel　　]

- Which VERSION of DynamixelSDK is it?

  [3.4.3]

- Any SOURCE SAMPLES?

  [ read_write.py from the SDK example]

 ISSUE:  Dynamixel MX-28 does not go to the exact goal position

For example, when running the read_write.py from the SDK example, if set goal position to 100, the returned position will be 95, 104, something very close but not the exact position.

Is there anyway to make the returned position same as the goal exact position? Wait longer time, or fine tune the PID parameters?

ROBOTIS-Leon, 2017-03-09:
Hi,

Sure,
how about fixing I gain :)

pguogithub, 2017-03-09:
ok, so just fine tune the PID #s. Thanks.

closed 2017-03-09 by ROBOTIS-Leon