ROBOTIS-GIT/DynamixelSDK #61: RX-24 F dynamixel servo connection with PC

Angit16, 2017-03-09:
Hi,

I have purchased a RX-24F dynamixel servo. I have USB-RS485 converter with me. I got to manage to power up the servo from a regulated supply. The red LED glowed for a moment and then turned off.
I used other dynamixel port for connecting with PC. 
GND->GND
TX+(RS-485)->D+(PIN3 of Dynamixel)
TX-(RS-485)->D-(PIN4 of Dynamixel)

Then I opened Dynamixel wizard and as provided in https://www.youtube.com/watch?v=RLnpps8eh6U&list=PLEf1s0tzVSnSgVzf4AREpat_P_HRLSiDn&index=2 , I searched for servo. But it was not found. Can anybody help me to identify the problem?

thanks,

ROBOTIS-Leon, 2017-03-09:
Hi,

the LED blinking shows only whether the power comes in, but not the communication status.

I can't guarantee the functionality if you use customized converter, but the pin connections are maybe not in problem.

Did you find any port that is connected and is shown through the Dynamixel wizard?

Angit16, 2017-03-09:
USB serial port (COM5) is showing for this particular USB and the same I selected in Dynamixel wizard.
I noticed that while searching only the TX LED of the converter is continuously blinking and the RX is always off.
thanks.

ROBOTIS-Leon, 2017-03-09:
Ah, sorry

I misunderstood the pin settings :<

I think that you are connecting a little bit wrong way, 

those should be connected like these:
Tx+ & Rx+ <-> D+
Tx- & Rx- <-> D- 
and some kind of control board that switches Tx or Rx should be used.

See <kbd>[e-manual](support.robotis.com/en/)</kbd> → <kbd>Product Information</kbd> → <kbd>Actuator</kbd> → <kbd>Dynamixel</kbd> → <kbd>MX Series</kbd>  → Connection to UART, and find the circuit diagram for UART2RS485 converter

Angit16, 2017-03-11:
Hi,

Thanks for your help.

I have made some modification in the connection.
Based on your link I did the following:
1. connected A(6) and B(7) of MAX485 to the D+ and D- respectively of the servo. 
2. RE of MAX485 made low (connected to GND) and DE is connected to VCC(+5V). 
3. RO and DI of MAX 485 are connected to RX1 and TX1 of novus USB-i485. TX- and RX- of it are connected to GND. MOD and GND are shorted.

As DE and RE are kept at constant high and low I expect to have transmit and receive to be enabled always. Then, I checked with the Dynamixel wizard.
But, still it does not work.

Please suggest how to fix this problem.

thanks,

ROBOTIS-Leon, 2017-03-11:
Okay,

It seems like you are going deeper than the `DynamixelSDK Issue` can handle. 

Basically, as the Issue form says, the Hardware issue can't be handled here, but I will pass this to hardware division. 

Please make inquiry sheet at <kbd>http://en.robotis.com/index/</kbd>→<kbd>Support</kbd>→<kbd>Troubleshooting</kbd> by copying and attaching:

```bash
Hi,

Thanks for your help.

I have made some modification in the connection.
Based on your link I did the following:

connected A(6) and B(7) of MAX485 to the D+ and D- respectively of the servo. Actually, I used a MAX485 module here
RE of MAX485 made low (connected to GND) and DE is connected to VCC(+5V).
RO and DI of MAX 485 are connected to RX1 and TX1 of novus USB-i485. TX- and RX- of it are connected to GND. MOD and GND are shorted.
As DE and RE are kept at constant high and low I expect to have transmit and receive to be enabled always. Then, I checked with the Dynamixel wizard.
But, still it does not work.

Please suggest how to fix this problem.

thanks,
```
, and you will get the response.
But make sure that 
it is not guaranteeded that the user can use their customized signal converter for the Dynamixel.

Angit16, 2017-03-11:
Thanks. I just post an inqury in ROBOTIS support.

ROBOTIS-Leon, 2017-03-11:
Great,

all the best

ROBOTIS-Leon, 2017-03-13:
Hey,
maybe you would get the answer faster by talking to cs@robotis.com

I just now knew that where I pointed is a little bit inappropriate location.
Sorry for the misleading.

closed 2017-03-19 by ROBOTIS-Leon