ROBOTIS-GIT/DynamixelSDK #62: dxl_monitor can't scan dynamixel pro, but sync_read_write can connect it

shelderzhang, 2017-03-10:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　 PRO M42-10-S260-R　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　USB2Dynamixel　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [3.4.3 ] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)
[image]
[image]

ROBOTIS-Leon, 2017-03-10:
Hi,

1) did you run `sudo chmod a+rw /dev/ttyUSBx`?
2) is your Dynamixel Baudrate set as 1000000 (1M) ?

shelderzhang, 2017-03-10:
Dynamixel_pro Baudrate is 115200, and I also can't find it using Robotplus in WIN7

ROBOTIS-Leon, 2017-03-10:
1. Make sure if the port number you selected in the ROBOPLUS is exactly the nunber which the USB2Dynamixel got. You can check in a simple way: plug the USB2Dynamixel out, then find which port number is gone. 
2. How did you find the PRO's baudrate? Does the OS except WIN7 can see your PRO? 
3. The example basically runs with baudrate @ 1M, and personally I recommend you to use your PRO at 1M. Change the baudrate as I said, and try it until it works. If it works, change again into 115200 (if you want) and edit the source of dxl_monitor to change the baudrate directly.

shelderzhang, 2017-03-12:
I have fixed this problem by recovering the dynamixel firmware  in the ROBOPLUS, thank you!

ROBOTIS-Leon, 2017-03-13:
Great,

I couldn't think about that way

closed 2017-03-13 by ROBOTIS-Leon