ROBOTIS-GIT/DynamixelSDK #65: sync_write_read fails to get present_position

shelderzhang, 2017-03-17:
Hi,
The example  sync_write_read  is always terminated due to fail to get present_position.
Note that : 
[TxRxResult] Incorrect status packet!
[ID:001] groupSyncRead getdata failed

How can I fix it ?

ROBOTIS-Leon, 2017-03-19:
Does read_write example work on the same environment?

shelderzhang, 2017-03-20:
yes, it works on the same environment. The error rate is high of the example

ROBOTIS-Leon, 2017-03-21:
Hi,

did you test sync_read_write example when the Dynamixels set as ID 1 ID 2 .. with Baudrate 1M?

I need more information to fix together, like a picture or video is good

ROBOTIS-Leon, 2017-03-30:
This issue will be closed since there was no actions within 7 days.
You can activate this issue whenever.

shelderzhang, 2017-04-12:
[image]
I haven't try it with baudrade 1M. I think the error occur in the code as shown below
[image]

I give the Dynamixel_PRO with a position trajectory that set goal point  every 0.02s (second). And when the  error is happen the Dynamixel_PRO will moves discontinuously

ROBOTIS-Leon, 2017-04-16:
@shelderzhang 

Hi, again

How many of Dynamixels do you control by sync read?

Did you try the number of it in minimum?

shelderzhang, 2017-04-18:
I just control two Dynamixels. I will try to control one of them

ROBOTIS-Leon, 2017-04-18:
@shelderzhang 

If you are using IndirectAddresses with Syncread, it might need more time intervals between each controls, because the communication packet would be much longer than normals.

shelderzhang, 2017-05-03:
It seems performance difference in different computers.  When I use the NUC(intel) to control the Dynamixels there are more errors. Even I just read it one time. ....... Is the baud rate too small

shelderzhang, 2017-05-03:
I'm so depressed in the experiments. It make the data with large noise

shelderzhang, 2017-05-03:
I think the error may be cased by the length of packet . In` Protocol2PacketHandler::txPacket()` It returns COMM_TX_ERROR, as following codes, 
``` 
  // check max packet length
  total_packet_length = DXL_MAKEWORD(txpacket[PKT_LENGTH_L], txpacket[PKT_LENGTH_H]) + 7;
  // 7: HEADER0 HEADER1 HEADER2 RESERVED ID LENGTH_L LENGTH_H
  if (total_packet_length > TXPACKET_MAX_LEN)
  {
    port->is_using_ = false;
    return COMM_TX_ERROR;
  }
```
The INDIRECTDATA information I use to control two Dynamixel_por like in indirect_address.cpp as following codes
```
// Control table address
#define ADDR_PRO_INDIRECTDATA_FOR_WRITE         634
#define ADDR_PRO_INDIRECTDATA_FOR_READ          638

// Data Byte Length
#define LEN_PRO_INDIRECTDATA_FOR_WRITE          4
#define LEN_PRO_INDIRECTDATA_FOR_READ           8

#define ADDR_PRO_INDIRECTADDRESS_FOR_WRITE    49                 
#define ADDR_PRO_INDIRECTADDRESS_FOR_READ    57   
```              
I write the GoalPos and read the Pos and Vel at the same time.

shelderzhang, 2017-05-05:
When I set time intervals between each controls as 0.01s, it  works well. But the interval time is too large for me. It  seems the txpacket is so busy...... It is strange that my PC don't have the bug

ROBOTIS-Leon, 2017-05-06:
@shelderzhang 

Hi,

Though I'm in holiday, I worked for this in my company since you are seemed urgent so far

I really hope that this finally makes you get into the goal

take a look the source and play video

it was filmed by using a M42-10-S260-R and H42-20-S300-R (bcs I didn't have any extras)

https://drive.google.com/drive/folders/0B2-t1GBeX7tuaUVsbTc1aDktb00?usp=sharing

the error shown at first scene was occurred by the voltage lackage (I used 12V but it needs 24V normally), and no error was appeared.

shelderzhang, 2017-05-11:
Thank you! There is no problem after I update the SDK

ROBOTIS-Leon, 2017-05-12:
@shelderzhang 

Oh, it is good for you

Good luck for your works

closed 2017-05-12 by ROBOTIS-Leon