ROBOTIS-GIT/DynamixelSDK #66: PID control using Dynamixel RX-24F

Angit16, 2017-03-19:
Hi,

I am looking forward to implement a PID control loop by writing it in C. Can I get the control over position and velocity by controlling the servo from PC?

thanks,

ROBOTIS-Leon, 2017-03-20:
you mean an example code?

Angit16, 2017-03-20:
yes, that would certainly help me. I would more likely look to learn about the RX-24 system, so that I can use a PID control for position and velocity correction.

ROBOTIS-Leon, 2017-03-21:
Hi,

Sorry, we don't have any example codes released.

If you are suffering on simultaneous multiple item control, 

just think about changing "LED control" to "Velocity control" in indirect_address.c

ROBOTIS-Leon, 2017-03-30:
This issue will be closed since there was no actions within 7 days.
You can activate this issue whenever.

closed 2017-03-30 by ROBOTIS-Leon