ROBOTIS-GIT/DynamixelSDK #67: Inconvenient way of getting meaning of packet result and error value

Toreno96, 2017-03-22:
I found out `packet_handler::printTxRxResult` and `packet_handler::printRxPacketError` to be a very inconvenient way of getting meaning of communication result and hardware error values, because it's enforcing the user what to do with that meaning and when, by printing it to a standard output at the moment of the call.

In my opinion, the better idea would be to have functions, which would return a proper string instead. The original ones could be kept for backward compatibility, but internally they could use the new ones.

I could even volunteer to implement them. After all, it's only a little modification to existing algorithm.

ROBOTIS-Leon, 2017-03-23:
@Toreno96 

Hi,

Thank you for your will of contribution.
There is <kbd>develop</kbd> branch in the repository.
Please make your pull request to the <kbd>develop</kbd> branch so that I can see the idea.

ROBOTIS-Leon, 2017-03-30:
@Toreno96 

This issue was solved by the pull request #67

ROBOTIS-Leon, 2017-03-30:
I'll check whether there is another examples with another languages which needs to be done as same,
and will close the issue.

Toreno96, 2017-08-18:
@ROBOTIS-Leon 

I saw that you deprecated `packet_handler::printTxRxResult` and `packet_handler::printRxPacketError`. Did you thought about using [[[deprecated]] attribute](http://en.cppreference.com/w/cpp/language/attributes) instead of runtime output? Or dependency on C++14 is unacceptable?

ROBOTIS-Leon, 2017-08-21:
@Toreno96 

Hi, again

Thank you for pointing out.

Yes, obviously, the latter.
Do you have any ideas?

closed 2017-08-21 by ROBOTIS-Leon

Toreno96, 2017-08-21:
@ROBOTIS-Leon 

Hi :)

What ideas do you mean? Alternative to runtime output about deprecation and [[deprecated]] attribute?

ROBOTIS-Leon, 2017-08-22:
@Toreno96 

Yes, if you have

Toreno96, 2017-10-17:
@ROBOTIS-Leon 

First of all, sorry for long silence, last few months were rather busy for me.

I found [this](https://stackoverflow.com/questions/295120/c-mark-as-deprecated/21265197#21265197) for simulating `[[deprecated]]` attribute when C++14 isn't available.

Additionally, C++14's solution could be accessible, too, because it is possible to check if selected feature is supported during compilation - via [__cplusplus macro](http://en.cppreference.com/w/cpp/preprocessor/replace) or via compiler vendor and version (much more reliable solution, but at the price of more time-consuming and tedious implementation).