ROBOTIS-GIT/DynamixelSDK #69: DYNAMIXEL SDK failed to read Moving flag

pguogithub, 2017-04-05:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  MX-28T

- Which CONTROLLER is it?

  USB2Dynamixel

- Which VERSION of DynamixelSDK is it?

 DynamixelSDK-3.4.3 for Windows

- Any SOURCE SAMPLES?

  Used the DynamixelSDK-3.4.3->Python->protocol1_0->read_write.py

My question:

Everything is working ok when I ran the read_write.py.

Then I tried to use read1ByteTxRx(portNum, 1, ID, 46) to  check if the  Goal position command execution is completed or not. According to MX-28T motor control table, at address 46, the single byte should indicate the Moving flag as :
0  Goal position command execution is completed.
1  Goal position command execution is in progress.

As I check the returned value from read1ByteTxRx, the value is huge, not 0 or 1, anyone encountered similar problem and how did you solve this problem?

Thanks

ROBOTIS-Leon, 2017-04-13:
Hello, 

please try this

```python
# Read present position
dxl_present_position = dynamixel.read2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_PRESENT_POSITION)
if dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION) != COMM_SUCCESS:
    dynamixel.printTxRxResult(PROTOCOL_VERSION, dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION))
elif dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION) != 0:
    dynamixel.printRxPacketError(PROTOCOL_VERSION, dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION))

# Read moving status
dxl_moving = dynamixel.read1ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, 46)
if dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION) != COMM_SUCCESS:
    dynamixel.printTxRxResult(PROTOCOL_VERSION, dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION))
elif dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION) != 0:
    dynamixel.printRxPacketError(PROTOCOL_VERSION, dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION))

print("[ID:%03d] GoalPos:%03d  PresPos:%03d IsMoving:%03d" % (DXL_ID, dxl_goal_position[index], dxl_present_position, dxl_moving))
```

In my computer, there was no problem

pguogithub, 2017-04-16:
Thank you very much!

closed 2017-04-16 by pguogithub