ROBOTIS-GIT/DynamixelSDK #70: Wiki - 3.1 Source File: Folder tree diagram has protocol 2.0 misspelled as protocol 1.0 under example folder

Shreeyak, 2017-04-07:
Under the example folder, protocol 2.0 misspelled:
[image]

ROBOTIS-Leon, 2017-04-13:
@Shreeyak 

Hi, and thank you for contribution :)

What you pointed out is modified just now

Shreeyak, 2017-04-13:
Hey,
You're welcome! Glad to help. I also wanted to contribute some other minor
changes. Don't know if you'll get this mail if I reply to it. Ping me back
if you do?

I want there to be more comments in the source code so that people can
understand it.

ROBOTIS-Leon, 2017-04-14:
Of course, keep going

Shreeyak, 2017-04-14:
Yeah, so I'm working on porting this library to a custom SoC running a
version of linux.

   - So, was going through the source code and found a severe lack of
   comments, which made understanding the code a bit hard, given the extremely
   similar function names. So, I've been commenting the code for my own
   documentation and would soon like to submit those as a pull request.

   - Comments throughout the code will make it much simpler for anyone else
   going through the code to understand it. I would also like to add to the
   documentation a section for developers that explains each of the source
   files and their hierarchy.

   - I also find the extremely similar function names (which call each
   other) extremely confusing. Is there any chance of looking at re-naming
   said functions in the official git repo? I'll send examples soon.

ROBOTIS-Leon, 2017-04-16:
@Shreeyak 

You mean like intel edison board - yacto linux, for example?

```bash
   - So, was going through the source code and found a severe lack of
   comments, which made understanding the code a bit hard, given the extremely
   similar function names. So, I've been commenting the code for my own
   documentation and would soon like to submit those as a pull request.

   - Comments throughout the code will make it much simpler for anyone else
   going through the code to understand it. I would also like to add to the
   documentation a section for developers that explains each of the source
   files and their hierarchy.
```

Yeah, that's a good point. Since I should manage whole languages of DynamixelSDK, the comments must be written in a general ways. I'm glad to see how the source has been reflected by users, but be sure that you are working on the <kbd>develop</kbd> branch. pull requests to the master were rejected because it was hard to manage currently.

```bash
   - I also find the extremely similar function names (which call each
   other) extremely confusing. Is there any chance of looking at re-naming
   said functions in the official git repo? I'll send examples soon.
```

Of course, let me take a look your suggestions. I'll take it all or partly when some are against the rules of DynamixelSDK management, but anyway.

ROBOTIS-Leon, 2017-04-21:
Hi,

```bash
 - Yes, it's similar to the edison, in that it's an SoC, but it's not one of the models available in the consumer market, like edison or yolo. It's a proprietary design, running a stripped down android.
```

Well, I had tested the SDK in the Raspberry Pi under Snappy Ubuntu Core, so your board must have work with it.

``` bash
 - I'm still working on first getting this integrated into our build, so will start working on improving it once that is done. Before I make any changes, however, can you share the DynamixelSDK management rules? It will be much easier to work on code keeping the rules in mind.
```

If you want to just comment on the source about what every functions mean, theres no specific rules so show me an example source codes which has your comments, and let's work in the rest of the codes as same as it showed after my review.

If you want to change the APIs, you should show me every point to point, and will take time.

``` bash
 - Some examples, for confusing naming, are as below: 
The correlation between these functions is not clear at all and takes a lot of digging to figure out. Also, seems conflicting, eg: what is readTx1? If its only Tx, how can one read????
```

readTx does only Transmit the package with 'Read' instruction. It provides readTxRx but also provides readTx and readRx separately because they can be used in thread process, but normally the users will readTxRx which waits the returned package during designated time.

and the number 1 means the protocol: Dynamixel, as you know, works under communication protocol 1 or 2 or both (but not the same time) depended on the models.

If you mean that txPacket and readTx make confusion, actually readTxPacket looks much better, but we thought that theres no needs for doing it.

But, we have been monitoring the needs and will follow if there are.

Shreeyak, 2017-04-23:
Hi,
1. I have managed to get the source code compiled with my Hardware. Having
issues in my hardware in switching from read mode to write mode (we are
using a tri state buffer for half duplexing). I will get started on the
points you mentioned above once I have fully integrated the SDK into my
build.

2. I also now need to port the Dynamixel SDK onto a MCU. I am currently
using an STM32 Nucleo board for testing.
I have built an interrupts-based half-duplex UART library with rx/tx ring
buffers for the STM. Now I think all I need to to create a new platform
check (similar to _linux64) and write functions for setupport, readport,
writeport.
Can you suggest the best way to get started on porting the SDK to the MCU?
I would love to generate a generalized solution that can be merged into the
SDK and can be used by others.

3. Please do send the SDK management rules so I can write code that follows
the rules.

Thanks!

Hi,

ROBOTIS-Leon, 2017-04-23:
``` bash
1. I have managed to get the source code compiled with my Hardware. Having
issues in my hardware in switching from read mode to write mode (we are
using a tri state buffer for half duplexing). I will get started on the
points you mentioned above once I have fully integrated the SDK into my
build.
```
Good Luck.

``` bash
2. I also now need to port the Dynamixel SDK onto a MCU. I am currently
using an STM32 Nucleo board for testing.
I have built an interrupts-based half-duplex UART library with rx/tx ring
buffers for the STM. Now I think all I need to to create a new platform
check (similar to _linux64) and write functions for setupport, readport,
writeport.
Can you suggest the best way to get started on porting the SDK to the MCU?
I would love to generate a generalized solution that can be merged into the
SDK and can be used by others.
```
I suggest you to see how the DynamixelSDK can be used in Arduino Platform. 
System is slightly different then as you think, but I guess it will help you.

``` bash
3. Please do send the SDK management rules so I can write code that follows
the rules.
```

My team will talk about the open source manual sooner, so then I will notice you

Shreeyak, 2017-04-23:
I just went back to the github wiki and noticed that it is written "Arduino
(coming soon)".

Do you suggest I wait until your team implements it on an Arduino? Any way
I can pitch in?

ROBOTIS-Leon, 2017-04-24:
That means, we are now developing Arduino version SDK for all Arduino compatible boards.

What I linked is an Arduino version Dynamixel SDK only for OpenCR (it is our product but is compatible with Arduino), and I told you because it is not as same as SDK for MCU you mentioned but is as similar as that, and I thought that it will be a help for your work.

Shreeyak, 2017-04-24:
Hi!
I think there might've been an error. There is no link.

Is it in the opencr git repo?

ROBOTIS-Leon, 2017-04-24:
@Shreeyak 

yes, it is in there

Shreeyak, 2017-04-24:
Thanks, I'll check it out!!

ROBOTIS-Leon, 2017-04-24:
@Shreeyak 

GL

ROBOTIS-Leon, 2017-07-11:
@Shreeyak 

Hi, 

I'll soon provide DynamixelSDK for Arduino here.
Reopen this issue if you have interest.

closed 2017-07-11 by ROBOTIS-Leon

Shreeyak, 2017-07-13:
Definitely interested!  Glad to see such fast developments at Robotis.
Great work!