ROBOTIS-GIT/DynamixelSDK #71: Dynamixel wizard can't find the dynamixels using normal USB to RS485 converter

amirhm64, 2017-04-08:
I am using DFROBOT's (https://www.dfrobot.com/wiki/index.php/Multi_USB/RS232/RS485/TTL_Converter_(SKU:TEL0070) )multi converter to communicate with my PRO dynamixels but my M54-60-S250 doesn't want to respond and cannot be found via Dynamixel wizard. Others like M54-40-S250 are fine and can be connected. I have performed firmware recovery on M54-60 but didn't help.
- When i use USB2Dynamixel everything is fine

ROBOTIS-Leon, 2017-04-08:
@amirhm64 

Hi.
Due to the rule of this bulletin board, the <kbd>help wanted</kdb> label was attached on this issue. 
You can get solution from other users or cs@robotis.com.

ROBOTIS-Leon, 2017-04-16:
@amirhm64 

This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

and I think, #61 might be a help for you

closed 2017-04-16 by ROBOTIS-Leon