ROBOTIS-GIT/DynamixelSDK #72: When the Dynamixel_pro powered in the negative position, the position sensor error

shelderzhang, 2017-04-13:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　 -PRO M42-10-S260-R　] 

- Which CONTROLLER is it?

  [　PC via USB2Dynamixel　　] 

- Any PICTURES or VIDEOS?

[image]
[image]

when Dynamixel_pro powered in the positive position in 10414 it is ok. then  I move it to negative position -2185 and power off it. After  powered again its position changes to 260817. It seems something wrong with the position sensor

ROBOTIS-Leon, 2017-04-14:
Hi, again

Please try firmware update on Dynamixel by following described in <kbd> [E-Manual](support.robotis.com/en/)</kbd> - <kbd> Software Help </kbd> - <kbd> Robo Plus </kbd>

and check it again.

If it still doesn't work, send your PRO for repair to [Repair Center](http://en.robotis.com/index/service_04.php?tab=4)

shelderzhang, 2017-04-18:
oh, no

ROBOTIS-Leon, 2017-04-18:
@shelderzhang 

let me know any updates are available.

shelderzhang, 2017-04-20:
I have to send it for repair to Repair Center

ROBOTIS-Leon, 2017-04-21:
@shelderzhang 

Sorry for the inconvenience,

I already told this issue to the center.
We will bring your Dynamixel to your home asap

ROBOTIS-Leon, 2017-04-23:
@shelderzhang 

Hi,

I'll close this issue until hardware is fixed.

You can reopen this in anytime

closed 2017-04-23 by ROBOTIS-Leon