ROBOTIS-GIT/DynamixelSDK #78: Reading dynamixel moving status

pyrrhicpk, 2017-05-09:
Hi,

I am trying to control two AX18A and four MX106T dynamixels using the latest dynamixel sdk. I have the following two questions.

1) I am using the python sync write code to move the servos simultaneously. However I do not find any example code for sync read. Should I read them sequentially or with sync read? Do you have any example code for sync read?

2) The example code in sync write continuously reads a servo present position until it is less than a threshold from the goal position. In my scenario if the servo fails to reach that threshold state (due to heavy load), the while loop never ends causing a halt. I was thinking of breaking the servo read loop by reading the moving status of the servo. However I do not find any example code for python for reading servo moving status. Would it be a good idea to check the status of servos this way before issuing the next move command? Do you have any example code for reading the servo moving status?

Thanks

ROBOTIS-Leon, 2017-05-10:
@pyrrhicpk 

Hi,

```bash
1. I am using the python sync write code to move the servos simultaneously. However I do not find any example code for sync read. Should I read them sequentially or with sync read? Do you have any example code for sync read?
```

Both AX18A and MX106T would be controlled by using protocol1.0, and unfortunately, protocol1.0 doesn't support `sync read function` and `bulk write function`. So, as a result, to use sync read function, you should use other models of Dynamixel which uses protocol 2.0. However, as a matter of fact, MX106T was started supporting protocol 2.0 (see [details](http://support.robotis.com/en/product/actuator/dynamixel_pro/communication.htm)) so get upgrade its version.

```bash
2. The example code in sync write continuously reads a servo present position until it is less than a threshold from the goal position. In my scenario if the servo fails to reach that threshold state (due to heavy load), the while loop never ends causing a halt. I was thinking of breaking the servo read loop by reading the moving status of the servo. However I do not find any example code for python for reading servo moving status. Would it be a good idea to check the status of servos this way before issuing the next move command? Do you have any example code for reading the servo moving status?
```

`moving` value shows its status by depending on some conditions, and sometimes it will not show the value as you want (you can check this out). From my experience to say, `moving` value is also calculated by threshold method, so that example code makes users set the threshold value by themselves.

pyrrhicpk, 2017-05-13:
Thanks for the information.

closed 2017-05-13 by ROBOTIS-Leon