ROBOTIS-GIT/DynamixelSDK #79: AX12 USB2AX DynamixelSDK error "[txrxresult] there is no status packet" on Raspberry pi 3

ThomasCaspar, 2017-05-15:
Hello,
I come to you because I use USB2AX on a raspberry pi 3 to control 7 AX12. From time to time my program sends me the error "[txrxresult] there is no status packet" blocking me all the action of the AX12. 
I tried to change the baud rate without success. This error occurs when sending data.

Can you help me ?

ROBOTIS-Leon, 2017-05-16:
@ThomasCaspar 

Hi,

I need more details to figure it out.

and I guess I saw in the network that [someone (named as apollokit)](https://github.com/apollokit/DynamixelSDK/network) had forked and was working with porting the USB2AX to DynamixelSDK.

ThomasCaspar, 2017-05-16:
Hi,

My problem is when I send an instruction to AX12 with write2ByteTxRx, sometimes  I have error "[txrxresult] there is no status packet" and Ax12 no reacts. For solve this problem i send in loop the instruction while i have error,  but it's not good solution.

ROBOTIS-Leon, 2017-05-17:
@ThomasCaspar 

Hi,

Did you see other branches worked for USB2AX?

closed 2017-05-23 by ROBOTIS-Leon