ROBOTIS-GIT/DynamixelSDK #80: Significantly slower function calls after updating to sdk version 3.4.4

TJMolnar, 2017-05-17:
Hello,

I am using the ros implementation of the sdk to control several robotics platforms, setup as follows:
OS - Ubuntu 16.04
ROS - Kinetic
Serial - RS-485 (usb2dynamixel)
Dynamixels - various, see below

I have noticed that since the 3.4.4 update the SDK has become significantly slower when reading and writing to the dynamixels. 

For example, when reading 10 bytes from one XM430-W350-R servo's control table at 1Mbs on protocol 2.0, the call of the readTxRx method in the packet handler takes approximately 15ms to return, which is significantly slower than what was able to be achieved previously (approximately 3ms). The group_bulk_read function also has an unusually high running time as well. Currently I am able to bulk_read 6 bytes for 18 MX-106 dynamixels on protocol 1.0 at a maximum rate of 60Hz, where before the update I was able to achieve rates of 100-120Hz). In addition, attempting to roll back to sdk version 3.4.3 results in communication failures being returned by the sdk.

The platforms I am running on utilise up to 30 dynamixels on a single RS-485 bus and so this is a significant issue. In the changelog it was stated that the changes are a fix in response to the Ubuntu update 16.04.2. Is it possible to find out why this hotfix occured? 

Any advice would be much appreciated.

ROBOTIS-Leon, 2017-05-17:
@TJMolnar 

Hi, 

there was a change in kernel issue relatives, and many users told us about errors occurred so much since there was an upgrade to ubuntu 16.04.2. 

Then, I noticed that it can be fixed by making `return time delay` longer. If your computer is okay with original value `4 msec`, you can use it as well.

Though I could lead the users to make it by themselves, but the main issue was the upgrade of the Ubuntu, so I should do that.

If theres other ideas to figure this out, please let me know

ROBOTIS-Leon, 2017-07-04:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-07-04 by ROBOTIS-Leon

TJMolnar, 2017-09-20:
Hi, 

Long time between updates but I would like to reopen the issue as I think I've found the root of the problem. After looking into the issue progressively I think the problem may be in the latency value for the serial port. Before the ubuntu update 16.04.2 the default state of the FTDI drivers in linux was low latency mode (1ms) but the update changed the default state back to the standard (16ms) latency.

I've confirmed this by using setserial on the usb2dynamixel port to put the port back into the low latency mode, e.g: `setserial /dev/ttyUSB0 low_latency` Then, changing the `return time delay` back to 4 results in the slowdown disappearing.

I believe the fix will be to modify the serial code in port_handler_linux.cpp to set the port into low latency mode when it is opened. The latency value for the port is stored in `/sys/bus/usb-serial/devices/<portname>/latency_timer` so you can check that to verify the problem.