ROBOTIS-GIT/DynamixelSDK #81: Moving multiple servos using time

pyrrhicpk, 2017-05-17:
Hi,
I am controlling multiple MX 106T servos in a robotic arm using the dynamixel sdk and python example code. However servos with different goal positions finish their motion at different times. I want all the servos (with different goal positions) to complete their motion at the same time. Does the sdk provide any way to move the servos to reach their goal position in a specified time?

Thanks

ROBOTIS-Leon, 2017-05-18:
@pyrrhicpk 

Hi,

Dynamixels don't provide any function for reach time control, so should adjust speed of the motor while it is in position control mode.

ex)
Goal velocity = (Goal position - Present position) / desired reach time

pyrrhicpk, 2017-05-18:
Thanks for the information.

ROBOTIS-Leon, 2017-05-18:
:)

pyrrhicpk, 2017-05-23:
Hi,

The above approach works good when the servos are moved with goal acceleration =0. What would be the equation if I use a goal acceleration = 1?

Thanks

ROBOTIS-Leon, 2017-05-23:
@pyrrhicpk 

if Vgoal - Vzero >= Agoal x T_desired
  V = Vzero + Agoal x T

else
  Until Ta (= (Vgoal - Vzero) / Agoal)
    V = Vzero + Agoal x T
  Since Ta
    V = Vgoal

pyrrhicpk, 2017-05-23:
Thanks for the info, but I am a bit confused with the variables.
Lets say I want to move a servo to S=50 degrees with a goal acceleration of 1 in T=5 seconds. As per the MX106 control table, one unit point for goal acceleration is 8.583 degree/s^2.
So in the above equations, I assume that: 
Vzero=S/T
Agoal=8.583
T_desired=5
If V denotes the required velocity then, What is Vgoal? And what is Ta? I also do not get what you mean by the line Since Ta

Sorry for the confusions.
Thanks

ROBOTIS-Leon, 2017-05-24:
@pyrrhicpk 

Hi,

I'm very sorry for my insincere words.

Vzero : supposed initial velocity at Tzero
Tzero : when Dynamixel starts moving
Vgoal : Goal velocity input value
Agoal : Goal acceleration input value
Tdesired : either when your observation completed or when you want to make finish its rotation
Ta (= (Vgoal - Vzero) / Agoal) : when its velocity reached to goal velocity earlier than Tdesired

I hope these might help you

closed 2017-07-04 by ROBOTIS-Leon