ROBOTIS-GIT/DynamixelSDK #83: How to run Dynamixel MX-28T servo at different speed in Matlab

fxs587, 2017-05-18:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　MX-28T　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [USB2Dynamixel　　　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [3.4.4] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [](　Link the SOURCE SAMPLES here, if necessary　)
Hi everybody, I am able to run my dynamixel mx-28t servo motor using Matlab simple read-write example inside SDK file. My question is that in my project, I will need to use 2 servo motor and I have to be able to change their speed and rotation separately. I don't know how to change the speed for each motor. Also, when I connect 2 servo motor together, they operate simultaneously which I wanna operate them separately. I am wondering does anyone help me?
Thanks a lot in advance.

ROBOTIS-Leon, 2017-05-19:
@fxs587 

Hi,

I understood what you told as you want to change goal position and goal velocity at the same time with MX?

fxs587, 2017-05-19:
Hi Leon
 I wanna change the goal position and goal velocity at different times for each servo  MX.  I connected two servos with the same USB dynamixel. I am just able to change the goal rotation and they rotate simultaneously which I don't want. Because I wanna have control to operate the servos with different goal position and velocity at different times.

ROBOTIS-Leon, 2017-05-23:
@fxs587 

Hi, 

then you should use `read_write` example.

just copy and paste readNbyteTxRx and writeNByteTxRx functions couple of times so that it is able to move multiple motors.

fxs587, 2017-05-23:
Hi Leon
Thank you so much. How about changing speed? How can I change the speed of each motor?

ROBOTIS-Leon, 2017-05-23:
@fxs587

Hi again,

the `read_write` example shows changing its `goal position` value, and you may find something starts with `ADDR_` which means the value of the `Control table address` of the Dynamixel you are using with.
So if you change it into other values like the address of `Goal velocity`, which can be referred to [ROBOTIS E-Manual](http://support.robotis.com/) - <kbd>Product Information</kbd> - <kbd>Actuator</kbd> - <kbd>Dynamixel</kbd> - <kbd>MX Series</kbd> - <kbd>MX-28AT</kbd> - Control Table Moving Speed, it will move as you want.

GrigoriyAY, 2017-05-29:
Hi fxs587,

I used with command for dynamixel library:
calllib('dynamixel', 'dxl_write_word',ID,32,SpeedValue)

closed 2017-07-04 by ROBOTIS-Leon