ROBOTIS-GIT/DynamixelSDK #84: Errors while reading servo position: [TxRxResult] There is no status packet! [TxRxResult] Incorrect status packet!

pyrrhicpk, 2017-05-19:
Hi,

I am controlling multiple MX106T MX64 and AX18 servos using the dynamixel sdk (protocol 1.0) and getting an error while reading the present position of the servos. There are 7 servos connected in a daisy chain fashion. For moving to goal positions, I am using the groupSyncWriteTxPacket example from SDK to move all the servos simultaneously. Immediately after the write command, I am reading the present position of each servo **individually** using the following code. 

while 1:
        # Read present position
        dxl_present_position = dynamixel.read2ByteTxRx(port_num, PROTOCOL_VERSION, DXL_ID, ADDR_MX_PRESENT_POSITION)
        if dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION) != COMM_SUCCESS:
            dynamixel.printTxRxResult(PROTOCOL_VERSION, dynamixel.getLastTxRxResult(port_num, PROTOCOL_VERSION))
        elif dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION) != 0:
            dynamixel.printRxPacketError(PROTOCOL_VERSION, dynamixel.getLastRxPacketError(port_num, PROTOCOL_VERSION))

However, occasionally, I get the following errors originating from **dynamixel.getLastTxRxResult** for 1 or 2 servos.

[TxRxResult] There is no status packet!
[TxRxResult] Incorrect status packet!

What are these errors and how can I possibly fix them?

Thanks

ROBOTIS-Leon, 2017-05-19:
@pyrrhicpk 

Hi,

it is actually possible when your loop runs quickly and so the communication speed can't follow.

So please bet to make it slower by attaching some delay()s or use faster baudrate or expand the return time delay, and try bulkread function which may act like syncread when you use MXs.

ROBOTIS-Leon, 2017-07-04:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

closed 2017-07-04 by ROBOTIS-Leon

meetndoshi, 2020-08-07:
I am facing a similar error, is there anyway to fix this? I have tried adding delays