ROBOTIS-GIT/DynamixelSDK #85: Problem communicating with XH430-W350-T using MATLAB Simulink through the Raspberry Pi on TxRx GPIO pins.

andrewmcconnell, 2017-05-22:
Hi, 
I have a XH430-W350-T. I am trying to communicate with this servo using MATLAB Simulink through the Raspberry Pi on TxRx GPIO pins. 
I have built the half-duplex level shift circuit suggested by Robotis.
I have written MATLAB code to assemble an instruction packet, and set GPIO pin 4 Hi to enable Tx comms.
The packet is transmitted ok, signal data appears on the DATA pin at the XH430.
 (See attached image SDS00001.BMP)
What is strange is that the signal from the Raspberry Pi 3 seems to be transmitted at 115200 Baud, although it is set at 57600.  (This might be an issue with set-up of the Siglent oscilloscope.)
The XH430 usually responds with a status packet after 1ms, which I can see on the DATA line. It seems to be at the same Baudrate.
 
But the problem is that the MATLAB program running on a PC is too slow to instruct the GPIO pin 4 “TX enable” to go low.
See SDS00002.BMP
MATLAB code:
for loop=1:300
    writeDigitalPin(rpiDyna,4,true) % Tx_enable
    write(DynaXH,uipkt) % write the instruction Packet
    writeDigitalPin(rpiDyna,4,false)
end
 
There is 30ms between each instruction, which is too slow to change GPIO tx enable to LO, to half duplex receive mode.
1. Is there a function on the Raspberry Pi that can handle the half duplex comms on the GPIO pins faster?
2. Can such a GPIO UART comms function be Modelled in Simulink and deployed to the Raspberry Pi?
--------------------------------------

- Which DYNAMIXEL is it?

  [　XH430-W350-T]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [MATLAB via Raspberry Pi] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [NONE] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [NA](　Link the SOURCE SAMPLES here, if necessary　)
[OscilloscopeImg.zip](https://github.com/ROBOTIS-GIT/DynamixelSDK/files/1018331/OscilloscopeImg.zip)

ROBOTIS-Leon, 2017-05-23:
@andrewmcconnell 

Hi,

what you are scoping at is a little far from what here can handle.
I'll leave this issue as opened for a while to get any other responses from the others, but I suggest you have a time to contact `Raspberry Pi` guys and `cs@robotis.com` for full-dimensional supports.

and in my experience to say, raspberry pi GPIO can only handle with baudrate at 115200 in stable. Of course you can change to another one but they were not stable enough to get communication with Dynamixel. 

In addition, me and the Mathworks are going to kickoff `Official DynamixelSDK for Matlab` which supports simulink as well and other designated stuffs, so stay tuned for updates.

andrewmcconnell, 2017-05-24:
Thanks Leon.
I'll keep digging, looking forward to the MATLAB collaboration.

ROBOTIS-Leon, 2017-05-24:
@andrewmcconnell 

GL

andrewmcconnell, 2017-06-04:
Hi All, 
Currently progressed simply by replacing the RPi with a CM9.04 and using the examples from the (Arduino-ish)  CM9.04 IDE. It works, and the "Learnings" document from the Robitis e-Manual has a few helpful hints. So now I can read the EEPROM values:
modelNumber: 232
modelInfo: 0
firmwareVersion: 38
dynamixelID: 1
baudRate: 1
responseDelayTime: 250
driveMode: 0
operatingMode: 3
secondaryID: 255
protocolVer: 2
homePosOffset: 0
vthreshMovDetect: 10
maxIntTempLim: 80
maxVoltLimit: 160
minVoltLimit: 95
maxPWMLimit: 117
maxCurrentLimit: 136
maxAccelerationLimit: 255
maxVelocityLimit: 24
maxPositionLimit: 255
minPositionLimit: 0
shutdownDynamixel: 52

And blink the LED from a loop().
All the examples there are written about 2012 however, so still figuring out the changes for each program to upgrade to XH430. It seems c functions like Dxl.getResult() no longer work.
Can't get movement by writing to Goal position yet. Have set #define GOAL_POSITION 116
and  Dxl.writeWord(ID_NUM, GOAL_POSITION, 300); but no success.
Is there a newer set of CM9.04 examples for the XH430?

ROBOTIS-Leon, 2017-07-04:
Sorry for the late response

DynamixelSDK for OpenCM will be released in end of August.

Since theres no official solution for X series in OpenCM, I recommend you to use upcoming new service.

ROBOTIS-Leon, 2017-07-11:
@andrewmcconnell 

Hi again,

see what's going on in develop-mid branch.

Thanks a lot for your patience.

ROBOTIS-Leon, 2018-04-09:
@andrewmcconnell

How are you,

this issue have been wanting a help from long time ago, 
but still no action is shown. I leave the question label,
but want an understanding of closing this issue. 

We will reopen the issue if any actions occur.

closed 2018-04-09 by ROBOTIS-Leon

fadhlian, 2020-08-08:
hi all @LeonJung and @andrewmcconnell any further clues for using raspberry pi in Simulink to control dynamixel servo?