ROBOTIS-GIT/DynamixelSDK #88: Installing both C and C++ bindings system wide.

de-vri-es, 2017-05-23:
Both the C and C++ headers have exactly the same names but different contents. That makes it cumbersome to install both the C and C++ headers on the same system.

Right now I have to install the headers for both packages in different subdirectories. For example, I used:
```
/usr/include/dynamixel_sdk_c
/usr/include/dynamixel_sdk_cpp
```

That works for me, and I don't mind having to do:
```
#include <dynamixel_sdk_cpp/dynamixel_sdk.h>
```

However, this isn't standard, so writing portable code or cmake/autoconf modules that work for both C and C++ out of the box with the provided install targets isn't really possible.

It would be great if the default install targets for C and C++ can coexist on the same system with the same prefix (`INSTALL_ROOT`). That could be achieved for example by renaming the C++ header to `dynamixel_sdk_cpp.h` (or even better in my view: `dynamixel_sdk.hpp`) and putting all the other C++ headers in `dynamixel_sdk_cpp(_linux,_window)` folders.

ROBOTIS-Leon, 2017-05-23:
@de-vri-es 

Hi,

What points out is seemed very horrible. I've never thought kinda that situation.
Thanks a lot.
Btw, did you see it coincidentally when you tried to use both of those SDKs in different language sequentially?
I had just thought in my work days that they don't need to be coexisted in all cases?

de-vri-es, 2017-05-23:
I encountered this when packaging the SDK for the Arch User Repository: https://aur.archlinux.org/packages/dynamixel-sdk/

Personally I want to use the C++ API, but I have to take into account other use cases as well when packaging. If they can't coexist they can't be packaged for a package manager. Plus I may need to compile tools using the C API while writing my own tools in C++

ROBOTIS-Leon, 2017-05-23:
@de-vri-es 

Aha, now it makes sense,
but let me think about it, since I heard that most of library files built by the other SDK or tools have been installed into the root in common way. 
Plus, the next version update will be carried out after June 20th, so care with your source by your repo temporarily.

Thanks for all

closed 2017-07-11 by ROBOTIS-Leon