ROBOTIS-GIT/DynamixelSDK #92: Unable to establish communication with  dynamexil M54-40-S250-R

jrgnicho, 2017-05-24:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　 M54-40-S250-R　]  

- Which CONTROLLER is it?

  [USB-to-Dynamixel dongle]

- Which VERSION of DynamixelSDK is it?

  whichever that's packaged with ros kinetic debian, I installed with the command below　　
```
sudo apt-get install ros-kinetic-dynamixel-workbench
```

- Error Message
```
core service [/rosout] found
process[single_manager_tutorial-1]: started with pid [29146]
[ INFO]: Succeeded to open the port(/dev/ttyUSB0)!;
[ INFO]: Succeeded to change the baudrate(57600)!;
[ INFO]: Scan Dynamixel Using Protocol 1.0;
[ INFO]: Scan Dynamixel Using Protocol 2.0;
[ERROR]: ...Failed to find dynamixel!;
================================================================================REQUIRED process [single_manager_tutorial-1] has died!
process has died [pid 29146, exit code -11, cmd /opt/ros/kinetic/lib/dynamixel_workbench_single_manager/dynamixel_workbench_single_manager __name:=single_manager_tutorial __log:=/home/ros-industrial/.ros/log/087512b6-40b3-11e7-a53a-a0c5891413c4/single_manager_tutorial-1.log].
log file: /home/ros-industrial/.ros/log/087512b6-40b3-11e7-a53a-a0c5891413c4/single_manager_tutorial-1*.log
Initiating shutdown!
```

- Launch File
Below is the launch file I'm running
```
<?xml version="1.0"?>
<launch>
  <arg name="device_name" default="/dev/ttyUSB0"/>
  <arg name="baud_rate" default="57600"/>

  <param name="device_name"      value="$(arg device_name)"/>
  <param name="baud_rate"        value="$(arg baud_rate)"/>

  <node name="single_manager_tutorial" pkg="dynamixel_workbench_single_manager" type="dynamixel_workbench_single_manager" required="true" output="screen"/>
</launch>
```

- Hardware setup description
  - linux machine runnging ubuntu 16.04 and ros-kinetic. 
  - I have an external power 24V/5Amps power supply hooked up to it for power and the 
  - USB-to-Dynamixel dongle for data.  

- Other Observations
When I first power up the motor the green LED lights up for a second and then it turns off for good.

routiful, 2017-05-25:
Hi :)

I will check it your issue as soon as possible.

However, Your question is about dynamixel_workbench. Next time, if you have any question about it, please use [dynamixel_workbench issues ](https://github.com/ROBOTIS-GIT/dynamixel-workbench/issues) page.

Thanks

jrgnicho, 2017-05-25:
I was able to resolve this by using the data port on the right side (back view of the actuator).  I'm not sure why it didn't work with the left data port.  This issue can probably be closed now, thanks

closed 2017-05-25 by ROBOTIS-Leon