ROBOTIS-GIT/DynamixelSDK #93: Make Dynamixel SDK installable for Python using "pip"

brettf, 2017-06-06:
Please, please make a Dynamixel SDK available via `pip`. Specifically I want to install the dependency upon Dynamixel by executing a command like: `pip install dynamixel`.

Of if you want separate distributions for different families then a collection of installs like:
```bash
pip install dynamixel-core
pip install dynamixel-servo-<family_name>
..etc..
```
..anything really that doesn't require that I parse and edit files in order to compile. 

- Which DYNAMIXEL is it?
  [　All　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?
  [　USB2Dynamixel　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?
  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)
  [　3.4.3+　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any SOURCE SAMPLES?
  I am trying to distribute [Servode](https://github.com/brettf/servode) via `pip`. As is, the Dynamixel SDK makes it pretty much impossible to package and distribute my own library to leverage the SDK without forking or hacking my way through.

ROBOTIS-Leon, 2017-06-07:
@brettf 

Hi,

Let me check this out, but it will take few weeks. 

Thank you.

brettf, 2017-06-07:
Just the fact that you're checking it out is excellent. Thanks. I've had to hack things together just to get my use of the Dynamixel SDK working in a straightforward, python-scriptable fashion. I can share the "hacked" code if you'd like.

ROBOTIS-Leon, 2017-06-07:
@brettf 

Please let me have a look

thanks

ROBOTIS-Leon, 2017-06-07:
@brettf 

and if you want to share via PR, pull at develop branch, not master branch

ROBOTIS-Leon, 2017-07-04:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.

brettf, 2017-07-06:
I now have the hacked code that I created to download and process the Dynamixel SDK, published and public [here](https://github.com/awslabs/aws-greengrass-mini-fulfillment/blob/master/groups/servo_setup.py)

Instead of this I just want to execute `pip install ...`

ROBOTIS-Leon, 2017-07-25:
@brettf 

Hi, again,

I had been having curiosity for pip since I first heard about the pip install. 
There must be so many friends who hope the pip installation as well as you.
At this point, I tried to upload the package into the pypi station, and failed, however.

Let me guess, you have an idea to get this well.
Any code modifications for uploading into pypi, or any PRs that you can upload, it doesn't matter.
Let me know or find the way how to put DynamixelSDK python into the pypi. 

The next DynamixelSDK release will happen in mid of August.

cod65, 2017-09-08:
.
+1 for pip.
I spent half a day installing/understanding pip before I realized (I think) that there is no pip-installable 'module' or 'package'. I'm tempted to start dragging files one-by-one into my Python install at C:\Python33\Lib\site-packages, but I have a feeling that this isn't going to work.

ROBOTIS-Leon, 2017-09-08:
@cod65 

As I know, there's a specific file structure that is recognizable by pip install.
Any knowledges available?

brettf, 2017-09-14:
@ROBOTIS-Leon these are the instructions for packaging code into PyPi that I have used successfully for multiple other projects. ([link](https://packaging.python.org/))

ROBOTIS-Leon, 2017-09-18:
@brettf 

Yeah, actually I saw this and tried, but it couldn't help me much

Anyway, I will retry this further after Oct.20th

Thank you for your kindness

closed 2017-11-28 by ROBOTIS-Leon