ROBOTIS-GIT/DynamixelSDK #94: Dynamixel SDK 3.4.5 in Visual Studio 2017 IDE

idrobot, 2017-06-09:
Hi there, one of my customers shares with us a situation while he used Dynamixel SDK 3.4.5 in Visual Studio 2017 IDE. He controls XM430-W350-R from PC. In VS2017 IDE, if you use getch()、kbhit() functions in your source codes, compile error will happen. You need to change to _getch()、_kbhit() so that compile will succeed.

I post here so that engineer can find out if it is true.

ROBOTIS-Leon, 2017-06-12:
@idrobot 

Hi,

This will be checked soon

ROBOTIS-Leon, 2017-06-14:
@idrobot 

Hi,

I just found that it works in VS 2017 with no modification. 
Me, too, got the message about deprecated function, 
but not always 'something deprecated' means 'must be changed' but means sometime 'suggest to change'.
It doesn't work absolutely?

Let me know about more detailed. 

Since the DynamixelSDK must be used by many old-version VS users, it is difficult to be following to latest systems everytime.

closed 2017-07-04 by ROBOTIS-Leon