ROBOTIS-GIT/DynamixelSDK #95: Additional features of the board OpenCM 9.04 (Reading data from pins, wireless connection)

GrigoriyAY, 2017-06-09:
Hello,
Could you tell me, how I can connect computer (Windows 10) with board OpenCM9.04 by using Bluetooth module HC-06 and MATLAB? And second question, I can control 3 Dynamixel motors (by using library "DynamixelTools"), but I didn't find information, how can I read data from pins via MATLAB. I would like to ensure the interaction of this board with force sensors (FSR400) and the ultrasonic sensor (HC-SR04), how can this be done?
Could you help me?

ROBOTIS-Leon, 2017-06-12:
@GrigoriyAY 

Hi,

I personally recommend you to have a talk with `cs@robotis.com` since this issue looks very far from what may be solved in this bulletin.

GrigoriyAY, 2017-06-13:
Ok, thank you.

ROBOTIS-Leon, 2017-07-04:
This issue will be closed since there was no actions for a while.
You can reopen this issue to show this issue to the users whenever.
The Label will be left as being attached.

closed 2017-07-04 by ROBOTIS-Leon