ROBOTIS-GIT/DynamixelSDK #98: SDK of LabVIEW

pjh3116, 2017-07-11:
Please fill out the questionaire to give you the best support service.  

- Which DYNAMIXEL is it?

  [　XM430-350R　]  ( ex: MX-64AT , XM430-350R , PRO H54-100-S500-R , ...)

- Which CONTROLLER is it?

  [　USB2Dynamixel　] ( ex: OpenCM , OpenCR , Arduino Uno , PC via USB2Dynamixel , ... )

- Which VERSION of DynamixelSDK is it?

  ([NOTE] DynamixelSDK Q&A is not available anymore if it is older than ver. 3.0.0, but feedback us)

  [　3.4.6　] ( ex: 3.4.2 , dxl_sdk_win32_v2_02.zip , ... )

- Any PICTURES or VIDEOS?

  [image]

- Any SOURCE SAMPLES?

  [no](　Link the SOURCE SAMPLES here, if necessary　)

Is dxl_x86_c.lvlib file coded in LabVIEW version 16?

I use the LabVIEW version 11, so i can't open example and register the library "dxl_x86_c.lvlib".

ROBOTIS-Leon, 2017-07-11:
@pjh3116 

Hi,

See [How to make libraries for Labview in Windows](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/3.2.7.1-LabVIEW-Windows).

Any additional questions are welcomed

pjh3116, 2017-07-11:
Thank you for kind answer.

I follow the manual well. (I have LabVIEW 2011 32bit)
But, after make libraries and run the some sub Vi. occur the error (below attach)

error : "Computer can't find VCRUNTIME140.dll"
[image]
error : "Fail, the library is disable to be found."
[image]

ROBOTIS-Leon, 2017-07-11:
@pjh3116 

Hi, again.

Basically, DynamixelSDK in the other languages use DLL file built by C language DynamixelSDK (you can find out the instructions [here](https://github.com/ROBOTIS-GIT/DynamixelSDK/wiki/3.2.1.1-C-Windows)), and so it needs VisualStudio. 

and the error message shown means that you need Visual Studio 2015 Redistributable version.

Please check this out:
http://devbible.tistory.com/430

ROBOTIS-Leon, 2017-07-24:
This issue will be closed since there was no actions for a while.
You can reopen this issue by telling me you are watching.

closed 2017-07-24 by ROBOTIS-Leon