ROBOTIS-GIT/DynamixelSDK #100: Maximum length of port name in linux c/c++ drivers

TJMolnar, 2017-07-19:
Hi, 

I have multiple serial ports on a single device and am trying to address them by their /dev/serial/by-id/ names instead of their /dev/ttyUSB symbolic links, so that they are always addressed correctly (as the /dev/ttyUSB symbolic links can switch if devices are plugged and unplugged). I noticed in port_handler_linux that the maximum length of port names is 30 however the port names I am intending to use are often 50-80 characters in length. Is it possible to increase or remove this limit?

Thanks.

ROBOTIS-Leon, 2017-07-19:
@TJMolnar 

Hi,

Thank you for the suggestion.
The next update will be solving those problems. Use your code until the update comes.

ROBOTIS-Leon, 2017-08-21:
@TJMolnar 

Hi, 
Just released DynamixelSDK ver. 3.5.1 is designed for solving your issue.

Check it and give me feedback

closed 2017-08-28 by ROBOTIS-Leon