ROBOTIS-GIT/DynamixelSDK #101: MX Series with Protocol 2.0?

ncalvr, 2017-07-25:
I've read on the support page and in the issues list here on GIT that since firmware v39 came out, the MX series are now compatible with Protocol 2.0. Unfortunately, there's no information or examples on how to implement/check this compatibility.   
  
Does this mean I can now just use the address table here: http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-106(2.0).htm  
and read/write to the MX series using Protocol 2.0 features like the position/velocity control?  
  
When I look at my MX series Dynamixels in R+ Manager 2.0 and Dynamixel Wizard 1.0.19.21, they only show the Protocol 1.0 control tables still. It's a bit confusing how you have said I can now utilise Protocol 2.0 using the SDK, but I can't then get feedback through R+ Manager or Dynamixel Wizard on whether or not my code has worked.  
  
I'm using the MX-R series (28, 64, 106) over a USB2Dynamixel?  
  
Cheers,  
Nick

ROBOTIS-Leon, 2017-07-25:
@ncalvr 

Hi,

since here is not appropriate bulletin that you can get the answer of your question, though I'll check this out and will put the answer here, but next time please use cs@robotis.com to get information about Dynamixel.

ncalvr, 2017-07-25:
@ROBOTIS-Leon  
  
I wasn't sure where to ask, as you say it does more refer to the Dynamixels, but I had run into the problem when using the SDK to write code for them with protocol 2.0.  Thanks for your help, though!

ROBOTIS-Leon, 2017-07-25:
@ncalvr 

Yeah, as I thought,
so I am finding out XD

ROBOTIS-Leon, 2017-07-28:
@ncalvr 

Hi, again

First of all, we will upload some video guide for MX series and Protocol 2.0 combination asap.
I will give you a link if the work is finished.

and here is my reply.

``` 
I've read on the support page and in the issues list here on GIT that since firmware v39 came out, the MX series are now compatible with Protocol 2.0. Unfortunately, there's no information or examples on how to implement/check this compatibility.
```

We will upload more details on the support page.

```
Does this mean I can now just use the address table here: http://support.robotis.com/en/product/actuator/dynamixel/mx_series/mx-106(2.0).htm
and read/write to the MX series using Protocol 2.0 features like the position+velocity control? That's what that page seems to imply. Or are Protocol 2.0 packets now just backwards compatible with the Protocol 1.0 MX series control table addresses when sending to MX series Dynamixels that are connected in series with Dynamixel PROs?
```
The answer is former one. You can use the Dynamixel MX as X series which supports more various control modes, though it could be used with position + velocity control in MX protocol 1.0 also, though. 

```
When I look at my MX series Dynamixels in R+ Manager 2.0 and Dynamixel Wizard 1.0.19.21, they only show the Protocol 1.0 control tables still. It's a bit confusing how you have said I can now utilise Protocol 2.0 using the SDK, but I can't then get feedback through R+ Manager or Dynamixel Wizard on whether or not my code has worked.

I'm using the MX-R series (28, 64, 106) over a USB2Dynamixel?
```

On the contrary to what you said, R+ Manager 2.0 will work with Dynamixel MX 2.0 but not MX 1.0 (technically, it can but will support in next update perhaps), so try update your R+ Manager 2.0 then let me guess what.

R+ Manager 1 Dynamixel Wizard will not work because, as I know, there are  no schedule for update

ncalvr, 2017-07-31:
@ROBOTIS-Leon 

Thanks so much for that. I heard back from cs@robotis.com and now that I've updated the firmware using R+ Manager 2.0, I can now see the Protocol 2.0 control tables. I'll put their response here for anyone searching for this info in future:  
  

> In order to change the protocol from 1.0 to 2.0, a firmware update or recovery is needed. In order to change your Dynamixel MX Series from Protocol 1.0 to Protocol 2.0, please follow the following steps:  
>1. Go to the Robotis US Software Download page and download [R+ Manager 2.0](http://www.robotis.us/roboplus2/)
>2. Connect your MX-Series Dynamixel to your computer through your USB2Dynamixel and power source  
>3. Select your Dynamixel under All Products and click Firmware Recovery. Steps for doing a firmware recovery are shown on our [e-Manual page](http://support.robotis.com/en/techsupport_eng.htm#software/roboplus2/r+manager2/r+manager2_explanation/recovering_the_firmware.htm) (Note that you must select the product you are viewing).  
>4. In order to switch back to 1.0, do a firmware recovery on RoboPlus Manager1.0

closed 2017-07-31 by ncalvr

ROBOTIS-Leon, 2017-07-31:
@ncalvr 

what a kindness

thank you so much

ROBOTIS-Leon, 2017-08-18:
@ncalvr 

Hi,

we just uploaded a video guide for MX series and Protocol 2.0 update.

ncalvr, 2017-08-21:
Thanks @ROBOTIS-Leon !