ROBOTIS-GIT/DynamixelSDK #106: Unable to open the serial port

ProjectFendi, 2017-08-16:
Hi everyone,

I'm using the dynamixel SDK on a raspberry pi 3 with Raspbian (latest available version), I follow all the instruction on the wiki (for the C library on linux)

All run clear except when I open the USB port, the ./read_write said "Failed to open the port !"

I think the problem is when I enter the command "sudo chdmod ax+rw /dev/tty/USB0", because on the raspberry, the USB address isn't like this (maybe ...)

Do you have any solution ?

Thanks a lot and have a good day

Best regards, Florian.

ProjectFendi, 2017-08-17:
I tried many baudrate (like 1000000 or 9600) and i tried to add some delays (in case if the the communication can't manage the speed of the loop ...)

But no changes, still the same message ...

ROBOTIS-Leon, 2017-08-18:
@FlorianRuen 

Hi,

I need more details for the solve. 
Devices you are using, which version of SDK you have downloaded, some views that can describe hardware connections.

What you attached here shows that the port are succeeded to be opened, but the Dynamixel couldn't respond to what you sent by some reason.

Can I guess that you have plugged the device into the GPIO port? or some external device?

FlorianRuen, 2017-08-18:
Hi,

I am using a raspberry pi 3 with the latest version of raspbian, and dynamixel XL320.

For the hardware I am using a USB2AX connected on usb port of the raspberry (/dev/ttyACM0)
this  usb2ax is plug on a hub to connect some dynamixel (like this https://www.generationrobots.com/en/400989-dynamixel-3pins-hub.html), but actually only one is connected.

The version of the dynamixel sdk is the latest on github, 3.4.7

When I am using the roboplus software, my dynamixel is ok and ready to receive some actions with the same connection between us.

Thanks for your help and have a good day !

ROBOTIS-Leon, 2017-08-18:
@FlorianRuen 

Hi,

unfortunately, we can't help you now with the device you have, as mentioned in the `Issue bulletin rules`

It should work, however, because there were many peoples who said about USB2AX and it worked.

but I know that many users are using USB2AX, so I'm thinking about testing with USB2AX here.

See other forks that tried with USB2AX or please wait until the test is done, but it will take much time.

FlorianRuen, 2017-08-18:
And directly with the GPIO do you think it can work ?

ROBOTIS-Leon, 2017-08-18:
@FlorianRuen

Nope, you need a board that changes uart signal into ttl or rs485 signal, and rpi basically transmits signal via gpio at 115200 baudrate, and no more.
actually you can follow some walkthrough to set its baudrate to over than 1M, but it is not stable in my experience

FlorianRuen, 2017-08-18:
Ok, I have a PIXL module from Roboticia i will try using this and I give you my feedback soon !

ROBOTIS-Leon, 2017-08-19:
@FlorianRuen 

Dynamixel SDK which includes `dxl_hal` are old ones, and is not supported now.
We only will deal with 3.x version.

btw, I've noticed that where you linked shows some minor change in `packet receive wait time` which means `timeout setting`. You can see it in paragraph here:

```
Gotchas with the Dynamixel SDK for Linux:

...
If you have frequent timeout errors, you might try to change l.154 of dxl_hal.c from gfRcvWaitTime = (float)(gfByteTransTime*(float)NumRcvByte + 5.0f); to gfRcvWaitTime = (float)(gfByteTransTime*(float)NumRcvByte + 34.0f);
...
```

So, I guess it may work if you change `LATENCY_TIMER` in [the source](https://github.com/ROBOTIS-GIT/DynamixelSDK/blob/master/c/src/dynamixel_sdk_linux/port_handler_linux.c) Line 46 to more that 8 msec.

Lets try like 30 or somethiing and give me feedback.

FlorianRuen, 2017-08-19:
The problem is very ladle ...

I change the LATENCY_TIMER like you said, the error message is the same, but i tried to use the dxl_monitor to scan all the dynamixel status, my motor is found.

But when i want to read present position, the error message is the same but different using each protocol (1.0 and 2.0).

What I do not understand is that it happens to find it (so by receiving a ping) but it can do nothing else ...

See the folowing screenshot: https://img4.hostingpics.net/pics/825616xxx.png

FlorianRuen, 2017-08-19:
Update: After an update of my USB2AX firmware, the commands run on the real motor (the motor move, the led color change) and I can read some data from it (like present position, current led color ...)

Last problem to solve: I have always the error message: "[RxPacketError] Hardware error occured. Check the error at Control Table ..."

Thanks for your help

ROBOTIS-Leon, 2017-08-20:
@FlorianRuen

You did great,

Thanks for a good reference

while running with roboplus manager, didnt you find any hardware alarm that will be shown in address 50?

FlorianRuen, 2017-08-20:
No error with RoboPlus after doing the same actions.

ROBOTIS-Leon, 2017-08-21:
@FlorianRuen 

1. If it occurs Hardware error, it will show non-zero value in Hardware Error Status @ ADDR 50. 
1.1 If it is non-zero, it will show same value of what is also shown @ ADDR 18 Alarm Shutdown.
1.2 See <kbd>http://support.robotis.com/en/</kbd> → <kbd>Product Information</kbd> → <kbd>Actuator</kbd> → <kbd>Dynamixel X</kbd> → <kbd>XL Series</kbd> → <kbd>XL320</kbd> → Control Table
1.3 XL320 will alarm when (1) input voltage is insufficient, (2) high temperature, (3) overload

If you think it is weird,
but if you could get read data,
remove the debug code of hardware error, and show me the error value in integer

FlorianRuen, 2017-08-22:
I will give you an answer tonight, but I think that the value in ADDR 50 it's 0 at this time

FlorianRuen, 2017-08-26:
Hi,

Sorry for the answer time, but I was very busy the last days.

I corrected the "Hardware error" problem, I think it's due to a bad contact in my USB2AX (or something like this)

Now, the script is correctly working with the USB2AX on Raspberry PI 3. Thanks a lot !

ROBOTIS-Leon, 2017-08-27:
@FlorianRuen 

Great work,

I should be ready for whom uses USB2AX someday

keep in touch

closed 2017-08-27 by ROBOTIS-Leon

yoggi56, 2019-08-25:
Hi!
I have the same issue but with AX-12+ servo.  Unfortenatelly firmware updating of USB2AX and changing the value of LATENCY_TIMER don't fix this problem.