ROBOTIS-GIT/DynamixelSDK #107: [C++] Risk of memory leak in PacketHandler::getPacketHandler and PortHandler::getPortHandler

Toreno96, 2017-08-17:
It's undocumented that pointers returned by methods mentioned in the title are owning pointers and memory that they points to should be manually deallocated, what cause a high risk of memory leaks in many applications, in my opinion.

While the user could rather easily find out about it in case of `PortHandler::getPortHandler` by reading implementation of this method, the same cannot be said about `PacketHandler::getPacketHandler`, where it's hidden a little "better".

My suggestion is to use smart pointers (because RAII) instead of raw pointers, specifically `std::unique_ptr` as it has little overhead and can easily and _efficiently_ convert to `std::shared_ptr` if shared ownership is needed, or at least document _very noticeably_ the risk of memory leak.

I could volunteer to implement the first solution, if it would be chosen :)

ROBOTIS-Leon, 2017-08-18:
@Toreno96 

Hi,
long time no see

What you pointed out must be a very good point, from my thought.

You know that systematical change should be done very carefully, and so it is slower than any other minor changes, even if it is very clear that it should be. 

Every PRs were not sure that it could be accepted, but I just hope to see the implementation. 

Today is the day of new version updates, and some branches will be removed on the list,
so please make PR from the develop branch which will be remade soon

Thanks.

Toreno96, 2017-10-17:
First of all, sorry for long silence, last few months were rather busy for me.

Before I will complete and show my implementation I have one important question: you told me [here](https://github.com/ROBOTIS-GIT/DynamixelSDK/issues/67) that C++14 cannot be used in SDK - but what about C++11?

ROBOTIS-Leon, 2017-11-30:
@Toreno96 

Hi, long time no see

I just uploaded the source which was changed as you said.
You can find it in develop branch now.
I think I should see if theres some negative effects on the other languages, such as matlab, labview, while c version sdk also uses the attribute deprecated

Toreno96, 2017-11-30:
@ROBOTIS-Leon 

No problem :)

But you know that `__attribute__((deprecated))` is keyword specific to GCC, and it would cause compiler error during build with Microsoft Visual C++?

Look again on the [page](https://stackoverflow.com/questions/295120/c-mark-as-deprecated/21265197#21265197) I linked, there is a snippet for making deprecation macro that will work on GCC _and_ Visual C++.

ROBOTIS-Leon, 2017-12-01:
@Toreno96 

Hi, again,

I reviewed all you attached, and changed it, tested in the other languages.
Only the thing left is whether it runs in Raspberry Pi

it seems like Matlab just regards the deprecated as none. the code just shows that it couldn't find the (deprecated) function XD

anyway, thanks for your help, and please give me any ideas if you have further.

and plus, I thought about the C++11 issue, and because of the complicated platform-dependency (we actually have a platform that uses gnu99 for compile), the project should be as it is now. When we change whole platform's compiler version later than 11, then I will change this. Is it okay for you? or.

Toreno96, 2017-12-01:
@ROBOTIS-Leon 

> and plus, I thought about the C++11 issue, and because of the complicated platform-dependency (we actually have a platform that uses gnu99 for compile), the project should be as it is now.

Without C++11 I don't have access to smart pointers from the standard library.

If you don't mind, I could use external implementation (e.g. Boost), but apart from the smart pointers I need to use a [static local variable](http://en.cppreference.com/w/cpp/language/storage_duration), which is not thread-safe until C++11. Would it be a problem?

ROBOTIS-Leon, 2017-12-04:
@Toreno96 

Hi, again,

I should find out anywhere that can't be used with C++11. 
If theres no problem, we can actually move the standard to C++11.

but it needs much time, so it is better that you firstly try to move the SDK on C++11

any ideas?

closed 2017-12-11 by ROBOTIS-Leon